qiboteam/qibocal #649: Protocols docs

Edoardo-Pedicillo, 2023-11-30:
This issue tracks the documentation of protocols.

- [ ]  Resonator Punchout / attenuation #644 
- [ ] Resonator spectroscopy / attenuation #575 
- [ ]  Qubit spectroscopy / qubit spectroscopy ef 
- [ ]  Rabi 
- [ ]  Ramsey
- [ ]  T1
- [ ]  T2
- [ ]  Zeno
- [ ]  Spin echo
- [ ]  Single-shot classification/ qutrit classification
- [ ]  All-XY
- [ ]  Ramsey detuned/ sequences
- [ ]  Dispersive shift / Dispersive shift qutrit
- [ ]  Flipping
- [ ]  Readout mitigation matrix
- [ ]  Coupler routines
- [ ]  Fast reset
- [ ]  Avoided Crossing
- [ ]  Qubit/resonator flux dependencies
- [ ]  Qubit/resonator crosstalk
- [ ]  Qubit flux tracking
- [ ]  RB
- [ ]  Readout optimization
- [ ]  TWPA calibration
- [ ]  Time of flight
- [ ]  Chevron
- [ ]  CZ virtualz
- [ ]  CHSH

andrea-pasquale, 2023-12-01:
Thanks @Edoardo-Pedicillo for opening the issue.
Here are some useful references:

- [Y. Gao, A practical guide for building superconducting quantum devices](https://arxiv.org/pdf/2106.06173.pdf)
- [P. Krantz, A Quantum Engineer’s Guide to Superconducting Qubits](https://arxiv.org/pdf/1904.06560.pdf)
- [M. Baur, Realizing Quantum Gates  and Algorithms with Three Superconducting Qubits](https://qudev.phys.ethz.ch/static/content/science/Documents/phd/PhD_Thesis_Baur_Matthias.pdf)
- [L. Ciociaro, Automatic Single Qubit Routines](https://qudev.phys.ethz.ch/static/content/science/Documents/Bachelor/Livio_Ciorciaro_BachelorThesis_2015.pdf)

Feel free to add other references.

rodolfocarobene, 2024-05-08:
Hi, I wanted to open an issue regarding documentation, but I see that there is already an open one. 
I will take care of this (at least partially... any help is welcomed). 

I was thinking that even if the protocol description is not very detailed, it's better than nothing. In particular, now it's not easy to understand exactly which all the protocols are and what are the possible parameters for them.

I think this is extremely important to extend the usability of qibocal, in particular when a stable release / publication is on sight. I will try to finish the PR before the end of next week (so that we have a ~ time-limit)

Edoardo-Pedicillo, 2024-05-16:
https://github.com/qiboteam/qibocal/pull/851#discussion_r1603263812

scarrazza, 2024-06-26:
Shall we close this?

andrea-pasquale, 2024-06-26:
> Shall we close this?

I think it is better to wait given that some protocols are missing.

closed 2024-10-29 by Edoardo-Pedicillo