qiboteam/qibocal #652: Customize release notes

andrea-pasquale, 2023-12-04:
I think that this could be useful given that the library is still quite unstable.

AleCandido, 2023-12-04:
A good guide:
https://keepachangelog.com/en/1.1.0/

there are even useful tools for doing this:
https://git-cliff.org/

andrea-pasquale, 2023-12-05:
It seems that 

> https://git-cliff.org/

works using [conventional commits](https://www.conventionalcommits.org/en/v1.0.0/) which I think that could be nice to use. Actually if we want to enforce them there is also a [pre-commit hook](https://github.com/compilerla/conventional-pre-commit), but perhaps this is too much...

The other option will be just to have a `CHANGELOG.md` and the author of the PR should be responsible for updating it.

closed 2025-02-14 by andrea-pasquale