qiboteam/qibocal #669: Backend in the protocols to reach execute_circuits()

Jacfomg, 2023-12-12:
It would be nice to have a way to reach the `execute_circuits()` to unroll circuit routines from the qibolab backend inside the protocols.

andrea-pasquale, 2023-12-13:
For this there should be two solutions:

1. (proper solution): Add `backend` in the protocols `acquisition`. Eventually we could replace backend with platform and access the platform using `backend.platform`.
2. (temporary solution): You can access the backend inside a routine by using `GlobalBackend` provided by qibo.

Jacfomg, 2023-12-13:
The 2nd solution works and it would look like what I was doing for https://github.com/qiboteam/qibocal/pull/597

Edoardo-Pedicillo, 2024-04-17:
After the merging of #597, we can close this issue.

closed 2024-04-17 by Edoardo-Pedicillo