qiboteam/qibocal #696: Document Ramsey experiment

hay-k, 2024-01-23:
We need to add proper documentation to all experiments, but let's start with this one.

Yesterday I learned that the Ramsey experiment is not doing true frequency detuning, but rather an artificial/fake detuning based on manipulating the relative phase of a pulse. This is not documented, and people don't even clearly remember the motivation behind. Before this is completely forgotten, it should be documented. 
- There should be brief documentation explaining what is the experiment doing at all in high level. 
- There should be clear mentions of particular choices (like the handling of detuning mentioned above) done in the experiment, their motivation, and any references if available.

Edoardo-Pedicillo, 2024-01-23:
> - There should be brief documentation explaining what is the experiment doing in high level. 

We are working in this direction, all our progresses about docs are documented in #649

andrea-pasquale, 2024-01-23:
Thanks @hay-k for opening the issue. 
Here is a reference to the artificial detuning by changing the relative phase
[image]
[Gao's paper](https://arxiv.org/pdf/2106.06173.pdf).
Theoretically the results should be the same but we should investigate more after running on hardware directly in both ways.
Moreover, this could be a good chance to use a proper notation instead of `n_osc` which corresponds to the number of oscillation that the user is expected to see. Perhaps indicating directly the detuning in Hz would be a better choice.

andrea-pasquale, 2024-06-21:
https://github.com/qiboteam/qibocal/pull/861#discussion_r1649014670

scarrazza, 2024-06-26:
Shall we close this?

andrea-pasquale, 2024-06-26:
I left this open on purpose because we want to expand a bit more the description of the Ramsey experiment.