qiboteam/qibocal #724: `Spin Echo` experiment slow

andrea-pasquale, 2024-02-20:
Currently `spin_echo` is quite slow given that the only implementation that we currently have relies on a `for` loop performed on software.
I believe that with the current version of qibolab it is not possible to perform this experiment with a sweeper given that the sequence is the following `RX90 - wait - RX - wait - RX90 - MZ` , correct @stavros11 @hay-k @alecandido?
For the time being I was thinking of providing the option to unroll the sequences.

If we all agree we could also decide to drop all protocols in qibocal that still perform this sort of for loop and just implement the unrolling. A possible idea would be to provide both unrolling and sweepers whenever possible like I'm doing in #721.

After benchmarking both unrolling and sweepers we could eventually decide to keep only one, but I believe that given that we support different drivers it is not a bad idea to keep both.

alecandido, 2024-02-20:
The idea for Qibolab 0.2 is to always fall back on unrolling, when sweepers are not available. So, (the moment that will become available) if you can implement with sweepers, do it.

The case for direct usage of unrolling should be when it is very complex to implement in the sweepers' language (like RB).

However, to do it in this way, we would need to deploy support at least for diagonal sweepers conversion (to unrolling).

andrea-pasquale, 2024-02-21:
I see, thanks for the summary.
If I understood correctly the qibocal user may not be able to choose the execution mode (`sweepers` or `unrolling`) since by default everything will be done with unrolling except in specific case where the sweeper can be used (?)

alecandido, 2024-02-21:
> If I understood correctly the qibocal user may not be able to choose the execution mode (`sweepers` or `unrolling`) since by default everything will be done with unrolling except in specific case where the sweeper can be used (?)

No, you will.

The idea is written here https://github.com/qiboteam/qibolab/issues/755:
```py
 def execute( 
     self, sequences: list[PulseSequence], options: ExecutionParameters, *sweepers: Sweeper 
 ): 
```
If you want to unroll, you will pass a list of sequences, that you produce before the call. But if you can specify with sweepers, it is recommended to do it anyhow, independently on the platform you're going to execute.

We can always do the conversion from sweepers to unrolling internally, just need to execute the loop in software (not during execution, but before, to generate the list).
But of course the opposite is impossible. So, whenever you can, you should use sweepers.

PR #725 (2024-02-27): Adding unrolling to spin echo
Closes #724.
I've added the option to unroll sequences when running spin echo which should improve performances.
There was also a typo since the protocol `spin_echo_sequences` was actually implementing `spin_echo_signal` which I fixed.

Checklist:
- [x] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
- [x] Compatibility with Qibo modules (Please edit this section if the current pull request is not compatible with the following branches).
    - [x] Qibo: `master`
    - [x] Qibo [... cut]
closed 2024-02-27 by andrea-pasquale