qiboteam/qibocal #732: Routines do not run on the specified qubits

stavros11, 2024-02-27:
If I try `qq acquire` and `qq report` on the following action runcard
```yaml
platform: dummy

qubits: [0]

actions:

  - id: resonator flux dependence
    priority: 0
    operation: resonator_flux
    parameters:
      freq_width: 10_000_000
      freq_step: 200_000
      bias_width: 0.6
      bias_step:  0.02
      nshots: 1024
      relaxation_time: 5000
```
I get a report that contains all five qubits of the dummy platform, while I would expect to get only qubit-0.

andrea-pasquale, 2024-02-27:
Thanks @stavros11 for opening the issue.
After merging https://github.com/qiboteam/qibocal/pull/682 we renamed `qubits` -> `targets`. The decision was mainly to have a notation that would be suitable for both qubits and qubit pairs, if you have a better suggestion than `targets` let me know.
Probably you are getting all 5 qubits because if you don't specify the `targets` entry `qibocal` will use as default all `qubits` available in the platform.
I tried to run your runcard using `targets` instead of `qubits` and it works as expected.

stavros11, 2024-02-27:
Thanks @andrea-pasquale, I missed this change. Indeed it works.

closed 2024-02-27 by stavros11