qiboteam/qibocal #736: dictionary changed size during iteration

JavierSerranoGarcia, 2024-02-28:
I am trying to run a punchout with the ZCU216 and spinq10q

```
platform: spinq10q_610_zcu216
targets: [6]
format: csv
actions:
  - id: resonator punchout
    priority: 0
    operation: resonator_punchout
    main: resonator_spectroscopy low power
    parameters:
      freq_width: 5_000_000
      freq_step: 1_000_000
      min_amp_factor: 0.02
      max_amp_factor: 2.4
      step_amp_factor: 0.1
      nshots: 100
      relaxation_time: 50
```
but I am getting a weird runtime error

`[Qibo 0.2.3|INFO|2024-02-28 12:16:58]: Loading platform /home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216
[Qibo 0.2.3|INFO|2024-02-28 12:16:58]: Using qibolab (/home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216) backend on /CPU:0
[Qibo 0.2.3|INFO|2024-02-28 12:16:58]: Loading platform /home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216
[Qibo 0.2.3|INFO|2024-02-28 12:16:58]: Using qibolab (/home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216) backend on /CPU:0
[33;20m[Qibocal 0.0.8|WARNING|2024-02-28 12:16:58]: Deleting previous directory reports/202402_punchout.[0m
[38;20m[Qibocal 0.0.8|INFO|2024-02-28 12:16:58]: Creating directory reports/202402_punchout.[0m
Traceback (most recent call last):
  File "/nfs/users/javier.serrano/qiboDev/env/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/cli/_base.py", line 72, in auto
    autocalibrate(runcard, folder, force, update)
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/cli/autocalibration.py", line 49, in autocalibrate
    executor = Executor.load(runcard, path, platform, runcard.targets, update)
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/auto/execute.py", line 53, in load
    graph=Graph.from_actions(card.actions),
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/auto/graph.py", line 35, in from_actions
    for task in dig.tasks():
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/auto/graph.py", line 68, in tasks
    for node, data in self.nodes.items():
  File "/usr/lib/python3.10/_collections_abc.py", line 910, in __iter__
    for key in self._mapping:
RuntimeError: dictionary changed size during iteration`

Any idea of what is going on?

andrea-pasquale, 2024-02-28:
Please remove this line 

```sh
  main: resonator_spectroscopy low power
```
if you add this line `qibocal` expect you to have a second protocol after the first one which will have the id `resonator_spectroscopy low power`, which is missing from your runcard.

JavierSerranoGarcia, 2024-02-29:
Thank you @andrea-pasquale

closed 2024-02-29 by JavierSerranoGarcia