qiboteam/qibocal #737: Could not broadcast input array from shape (100,) into shape ()

JavierSerranoGarcia, 2024-02-29:
I am finding this error executing this routine

```
platform: spinq10q_610_zcu216
targets: [6]
format: csv
actions:

  - id: resonator_spectroscopy high power
    priority: 0
    operation: resonator_spectroscopy
    # main: resonator punchout
    parameters:
      freq_width: 10_000_000
      freq_step: 100_000
      amplitude: 0.4
      power_level: high
      nshots: 1024
      relaxation_time: 0
```

and the error

`[Qibo 0.2.3|INFO|2024-02-29 10:36:25]: Loading platform /home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216
[Qibo 0.2.3|INFO|2024-02-29 10:36:25]: Using qibolab (/home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216) backend on /CPU:0
[Qibo 0.2.3|INFO|2024-02-29 10:36:25]: Loading platform /home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216
[Qibo 0.2.3|INFO|2024-02-29 10:36:25]: Using qibolab (/home/users/javier.serrano/qiboDev/qibolab_platforms_qrc/spinq10q_610_zcu216) backend on /CPU:0
[33;20m[Qibocal 0.0.8|WARNING|2024-02-29 10:36:25]: Deleting previous directory reports/202402_res_spec_high_6_7.[0m
[38;20m[Qibocal 0.0.8|INFO|2024-02-29 10:36:25]: Creating directory reports/202402_res_spec_high_6_7.[0m
[Qibo 0.2.3|INFO|2024-02-29 10:36:25]: Connecting to instrument <qibolab.instruments.rfsoc.driver.RFSoC object at 0x7ff6d76611e0>.
[Qibo 0.2.3|INFO|2024-02-29 10:36:25]: Connecting to instrument <qibolab.instruments.rohde_schwarz.SGS100A object at 0x7ff6d7661120>.
[38;20m[Qibocal 0.0.8|INFO|2024-02-29 10:36:25]: Executing mode autocalibration on resonator_spectroscopy high power iteration 0.[0m
[Qibo 0.2.3|INFO|2024-02-29 10:36:25]: Minimal execution time (sweep): 0.256
Connected to: Rohde&Schwarz SGS100A (serial:1416.0505k02/113300, firmware:4.2.76.0-4.30.046.295) in 0.19s
Traceback (most recent call last):
  File "/nfs/users/javier.serrano/qiboDev/env/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nfs/users/javier.serrano/qiboDev/env/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/cli/_base.py", line 72, in auto
    autocalibrate(runcard, folder, force, update)
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/cli/autocalibration.py", line 56, in autocalibrate
    for _ in executor.run(mode=ExecutionMode.autocalibration):
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/auto/execute.py", line 142, in run
    completed = task.run(
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/auto/task.py", line 124, in run
    completed.data, completed.data_time = operation.acquisition(
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/auto/operation.py", line 41, in wrapper
    out = func(*args, **kwds)
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/protocols/characterization/resonator_spectroscopy.py", line 174, in _acquisition
    data.register_qubit(
  File "/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/auto/operation.py", line 202, in register_qubit
    ar[key] = value
ValueError: could not broadcast input array from shape (100,) into shape ()`

andrea-pasquale, 2024-02-29:
Is this happening with the current main of both `qibolab` and `qibocal`?
It seems like an issue related to the shape of the results which is returned by qibolab/qibosoq.
In qibolab there is a test to check if the shapes returned by all drivers for different acquisitions are the same https://github.com/qiboteam/qibolab/blob/main/tests/test_result_shapes.py in the README you can check how to run this test.
If tests are failing probably we need to modify the rfsoc driver to make it behave like all others.

alecandido, 2024-03-01:
@andrea-pasquale do you understand what could be the `100`?

Because, even if it were the number of shots, that were not being correctly averaged or something like that, it would have been `1024`. So, having a guess for what the wrong `100` could be would help to trace it in the driver (assuming the problem is there).

alecandido, 2024-03-01:
Using the current `main` (both Qibolab and Qibocal) I can't reproduce the error, because I'm getting a different one:

```py
(env) I have no name!@ramhan:~$ qq auto spinq-javier-737.yml
[Qibo 0.2.4|INFO|2024-03-01 15:06:54]: Loading platform /nfs/users/alessandro.candido/qibolab_platforms_qrc/spinq10q_610_zcu216
[Qibo 0.2.4|INFO|2024-03-01 15:06:54]: Using qibolab (/nfs/users/alessandro.candido/qibolab_platforms_qrc/spinq10q_610_zcu216) backend on /CPU:0
[Qibo 0.2.4|INFO|2024-03-01 15:06:54]: Loading platform /nfs/users/alessandro.candido/qibolab_platforms_qrc/spinq10q_610_zcu216
[Qibo 0.2.4|INFO|2024-03-01 15:06:54]: Using qibolab (/nfs/users/alessandro.candido/qibolab_platforms_qrc/spinq10q_610_zcu216) backend on /CPU:0
[Qibocal 0.0.8|INFO|2024-03-01 15:06:54]: Directory 2024-03-01-000-alessandro-candido already exists.
[Qibocal 0.0.8|INFO|2024-03-01 15:06:54]: Trying to create directory 2024-03-01-001-alessandro-candido
[Qibocal 0.0.8|INFO|2024-03-01 15:06:54]: Creating directory 2024-03-01-001-alessandro-candido.
Traceback (most recent call last):
  File "/nfs/users/alessandro.candido/env/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/qibocal/cli/_base.py", line 72, in auto
    autocalibrate(runcard, folder, force, update)
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/qibocal/cli/autocalibration.py",line 49, in autocalibrate
    executor = Executor.load(runcard, path, platform, runcard.targets, update)
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/qibocal/auto/execute.py", line 53, in load
    graph=Graph.from_actions(card.actions),
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/qibocal/auto/graph.py", line 35,in from_actions
    for task in dig.tasks():
  File "/nfs/users/alessandro.candido/env/lib/python3.10/site-packages/qibocal/auto/graph.py", line 68,in tasks
    for node, data in self.nodes.items():
  File "/usr/lib/python3.10/_collections_abc.py", line 910, in __iter__
    for key in self._mapping:
RuntimeError: dictionary changed size during iteration
```
(of course `spinq-javier-737.yml` is the runcard above)

I'm trying to check if I can quickly overcome this issue...

alecandido, 2024-03-01:
I managed to reproduce the issue, using https://github.com/qiboteam/qibolab_platforms_qrc/tree/spinq10q_610_zcu216 (and in particular https://github.com/qiboteam/qibolab_platforms_qrc/commit/85125bfe6f89ddc8463297f6eabc8f992940a1ab).

I'm using the current `main` of both Qibolab and Qibocal.

GabrielePalazzo, 2024-03-01:
> @andrea-pasquale do you understand what could be the `100`?

It seems to be the number of different frequencies used (freq_width / freq_step).

andrea-pasquale, 2024-03-01:
Indeed, it was just a guess. 
I think the fastest way to debug should be to run https://github.com/qiboteam/qibolab/blob/main/tests/test_result_shapes.py with that platform.

alecandido, 2024-03-01:
I also tried with `iqm5q` (I only had to change the qubit ID), and I can confirm there is no similar issue, so it is definitely a drivers' problem.

alecandido, 2024-03-01:
> Random (not required) guess: qibocal is averaging an array of shape (because the driver is returning an array of incorrect shape) `(nfreqs*nshots,)` into a zero-dimensional number, but instead it should be an array of shape `(nfreqs,)`..

I also believe something like that, but it's actually the opposite: there is a missing average, because you're trying to set a 100-dimensional array as a single element.

andrea-pasquale, 2024-03-01:
So it could be that the rfsoc is returning the matrix transposed (?). We are just averaging over the wrong axis... (just a guess also)

alecandido, 2024-03-01:
> So it could be that the rfsoc is returning the matrix transposed (?). We are just averaging over the wrong axis... (just a guess also)

Not even that, unless you're averaging over a single element. Because it should be a single element, and it is `(100,)`.
If it were as you said, it would have been `(100,)`, but a request for `(1024,)` should have been reported.

GabrielePalazzo, 2024-03-01:
> So it could be that the rfsoc is returning the matrix transposed (?). We are just averaging over the wrong axis... (just a guess also)

It needs to be debugged for sure. As far as I rembember, the results over a single frequency are added together in qibolab (https://github.com/qiboteam/qibolab/blob/595351782d55a20fa2b8c35104c000092e42f585/src/qibolab/result.py#L114). If they have the wrong shape, the `np.append` behaves weirdly.

alecandido, 2024-03-01:
@andrea-pasquale the tests are actually useful!

```py
(env) I have no name!@ramhan:~/qibolab$ pytest tests/test_result_shapes.py -m qpu --platform spinq10q_610_zcu216
========================================= test session starts ==========================================
platform linux -- Python 3.10.12, pytest-8.0.2, pluggy-1.4.0
rootdir: /nfs/users/alessandro.candido/qibolab
configfile: pyproject.toml
plugins: dash-2.15.0, env-1.1.3, cov-4.1.0
collected 8 items

tests/test_result_shapes.py .F.F.F.F                                                             [100%]

=============================================== FAILURES ===============================================
_________________________________ test_discrimination_singleshot[True] _________________________________

connected_platform = Platform(name='/nfs/users/alessandro.candido/qibolab_platforms_qrc/spinq10q_610_zcu216', qubits={6: Qubit(name=6, bare... resonator_type='2D', couplers={}, is_connected=True, topology=<networkx.classes.graph.Graph object at 0x7fdeeb383610>)
sweep = True

    @pytest.mark.qpu
    @pytest.mark.parametrize("sweep", [False, True])
    def test_discrimination_singleshot(connected_platform, sweep):
        result = execute(
            connected_platform,
            AcquisitionType.DISCRIMINATION,
            AveragingMode.SINGLESHOT,
            sweep,
        )
        assert isinstance(result, SampleResults)
        if sweep:
>           assert result.samples.shape == (NSHOTS, NSWEEP1, NSWEEP2)
E           assert (2000,) == (50, 5, 8)
E
E             At index 0 diff: 2000 != 50
E             Right contains 2 more items, first extra item: 5
E             Use -v to get more diff

tests/test_result_shapes.py:54: AssertionError
----------------------------------------- Captured stderr call -----------------------------------------
[Qibo 0.2.4|INFO|2024-03-01 16:07:48]: Minimal execution time (sweep): 0.009080000000000001
------------------------------------------ Captured log call -------------------------------------------
INFO     qibo.config:platform.py:294 Minimal execution time (sweep): 0.009080000000000001
___________________________________ test_discrimination_cyclic[True] ___________________________________

connected_platform = Platform(name='/nfs/users/alessandro.candido/qibolab_platforms_qrc/spinq10q_610_zcu216', qubits={6: Qubit(name=6, bare... resonator_type='2D', couplers={}, is_connected=True, topology=<networkx.classes.graph.Graph object at 0x7fdeeb383610>)
sweep = True

    @pytest.mark.qpu
    @pytest.mark.parametrize("sweep", [False, True])
    def test_discrimination_cyclic(connected_platform, sweep):
        result = execute(
            connected_platform, AcquisitionType.DISCRIMINATION, AveragingMode.CYCLIC, sweep
        )
        assert isinstance(result, AveragedSampleResults)
        if sweep:
>           assert result.statistical_frequency.shape == (NSWEEP1, NSWEEP2)
E           assert (40,) == (5, 8)
E
E             At index 0 diff: 40 != 5
E             Right contains one more item: 8
E             Use -v to get more diff

tests/test_result_shapes.py:67: AssertionError
----------------------------------------- Captured stderr call -----------------------------------------
[Qibo 0.2.4|INFO|2024-03-01 16:07:50]: Minimal execution time (sweep): 0.009080000000000001
------------------------------------------ Captured log call -------------------------------------------
INFO     qibo.config:platform.py:294 Minimal execution time (sweep): 0.009080000000000001
__________________________________ test_integration_singleshot[True] ___________________________________

connected_platform = Platform(name='/nfs/users/alessandro.candido/qibolab_platforms_qrc/spinq10q_610_zcu216', qubits={6: Qubit(name=6, bare... resonator_type='2D', couplers={}, is_connected=True, topology=<networkx.classes.graph.Graph object at 0x7fdeeb383610>)
[... 36 lines cut ...]
E           assert (40,) == (5, 8)
E
E             At index 0 diff: 40 != 5
E             Right contains one more item: 8
E             Use -v to get more diff

tests/test_result_shapes.py:93: AssertionError
----------------------------------------- Captured stderr call -----------------------------------------
[Qibo 0.2.4|INFO|2024-03-01 16:07:52]: Minimal execution time (sweep): 0.009080000000000001
------------------------------------------ Captured log call -------------------------------------------
INFO     qibo.config:platform.py:294 Minimal execution time (sweep): 0.009080000000000001

---------- coverage: platform linux, python 3.10.12-final-0 ----------
Coverage HTML written to dir htmlcov
Coverage XML written to file coverage.xml

======================================= short test summary info ========================================
FAILED tests/test_result_shapes.py::test_discrimination_singleshot[True] - assert (2000,) == (50, 5, 8)
FAILED tests/test_result_shapes.py::test_discrimination_cyclic[True] - assert (40,) == (5, 8)
FAILED tests/test_result_shapes.py::test_integration_singleshot[True] - assert (2000,) == (50, 5, 8)
FAILED tests/test_result_shapes.py::test_integration_cyclic[True] - assert (40,) == (5, 8)
===================================== 4 failed, 4 passed in 8.96s ======================================
```

I can now use those to debug :)

andrea-pasquale, 2024-03-01:
You should thank @stavros11, he was the one who coded the test 😄

alecandido, 2024-03-01:
But first I should debug the tests, since they are failing also with `iqm5q` (but that seems trivial, it could be just a swap):

```py
FAILED tests/test_result_shapes.py::test_discrimination_singleshot[True] - assert (5, 50, 8) == (50, 5,8)
FAILED tests/test_result_shapes.py::test_discrimination_cyclic[False] - IndexError: list index out of range
FAILED tests/test_result_shapes.py::test_integration_singleshot[False] - IndexError: list index out of range
FAILED tests/test_result_shapes.py::test_integration_singleshot[True] - assert (5, 50, 8) == (50, 5, 8)
FAILED tests/test_result_shapes.py::test_integration_cyclic[False] - IndexError: list index out of range
===================================== 5 failed, 3 passed in 31.81s =====================================
```

andrea-pasquale, 2024-03-01:
Ok, but the tests that you showed are not related to the resonator spectroscopy. In the tests the sweep is over `bias` and `frequency`. 
https://github.com/qiboteam/qibolab/blob/595351782d55a20fa2b8c35104c000092e42f585/tests/test_result_shapes.py#L32-L37
For the spectroscopies we are just sweeping frequencies...
The protocols where we sweep both bias and frequency are the flux dependences, which up to know are performed without single_shot, which should explain why we don't see the problem in qibocal.
Maybe I pointed you to the wrong tests but at least we found something useful. We need to verify only what is shape of the output for the rfsoc driver when doing a single frequency sweep.

stavros11, 2024-03-01:
> But first I should debug the tests, since they are failing also with `iqm5q` (but that seems trivial, it could be just a swap):
> 
> ```python
> FAILED tests/test_result_shapes.py::test_discrimination_singleshot[True] - assert (5, 50, 8) == (50, 5,8)
> FAILED tests/test_result_shapes.py::test_discrimination_cyclic[False] - IndexError: list index out of range
> FAILED tests/test_result_shapes.py::test_integration_singleshot[False] - IndexError: list index out of range
> FAILED tests/test_result_shapes.py::test_integration_singleshot[True] - assert (5, 50, 8) == (50, 5, 8)
> FAILED tests/test_result_shapes.py::test_integration_cyclic[False] - IndexError: list index out of range
> ===================================== 5 failed, 3 passed in 31.81s =====================================
> ```

I remember we had this issue before but with @Jacfomg we fixed it in https://github.com/qiboteam/qibolab/pull/690 where these tests were introduced. Did it come back now? I am pretty sure it was fixed that time, otherwise we probably wouldn't merge the PR (if the introduced tests were not passing...).

alecandido, 2024-03-01:
> For the resonator/qubit spectroscopies we are just sweeping frequencies...

True

> The protocols where we sweep both bias and frequency are the flux dependences, which up to know are performed without single_shot, which should explain why we don't see the problem in qibocal.

It doesn't matter that much: `nshots` should always be outermost or innermost, and all the sweepers should be on the same side, in the same order you passed to the routine. It should not depend on the parameters' types at all, otherwise it's a mess.

> Maybe I pointed you to the wrong tests but at least we found something useful. We need to verify only what is the shape of the output for the rfsoc driver when doing a single frequency sweep using singleshot.

For sure, they were correct in the sense that this is a wrong behavior, in the same domain of the reported issue.
But even more: the QICK drivers are behaving differently from the Zurich ones. So, it could be exactly the issue. I'm pretty sure they were the correct tests (at least to start with).

alecandido, 2024-03-01:
> I remember we had this issue before but with @Jacfomg we fixed it in [qiboteam/qibolab#690](https://github.com/qiboteam/qibolab/pull/690) where these tests were introduced. Did it come back now? I am pretty sure it was fixed that time, otherwise we probably wouldn't merge the PR (if the introduced tests were not passing...).

I also had a deja vu, but I didn't remember exactly...

andrea-pasquale, 2024-04-04:
@rodolfocarobene can we close this issue?

rodolfocarobene, 2024-04-05:
There is a PR in qibolab that will solve the problem. It is waiting for review.
In any case, I would say that we can close this issue since the problem is not in qibocal

alecandido, 2024-04-05:
> In any case, I would say that we can close this issue since the problem is not in qibocal

Well, at a second thought: are we confident that Qibocal is always consistent with the shape handling?

Maybe it's worth a check.

closed 2024-04-16 by rodolfocarobene