qiboteam/qibocal #743: Add report failed information in the report.

andrea-pasquale, 2024-03-03:
Maybe we should write on the report that there has been an attempt, but it failed.

_Originally posted by @alecandido in https://github.com/qiboteam/qibolab_platforms_qrc/issues/119#issuecomment-1975158563_

andrea-pasquale, 2024-03-03:
Actually I already discussed this point with @hay-k when working on #707.
It might be non-trivial to do this given that missing `Results` could mean two different things:
1. Using `qq auto` and observing the fit failing.
2. Using `qq acquire` which means no fitting.

We might need to store the "execution mode" (`auto`, `acquire`, ...) to be able to discriminate the two different cases which I described above.

alecandido, 2024-03-03:
> We might need to store the "execution mode" (`auto`, `acquire`, ...) to be able to discriminate the two different cases which I described above.

I don't think so, I believe you could make it more isolated.

You could handle a failed fit, dumping a marker file for the failure (I believe we're talking about a "soft" fail, since a hard one, i.e. a crash, would immediately report an error to the user).
https://github.com/qiboteam/qibocal/blob/ddba2d3da995a308756ed25a1072b7e230a546ac/src/qibocal/auto/task.py#L227-L228

PR #745 (2024-03-12): Adding error message when fitting failed
Closes #743 .

The following implementation does not require a marker file which I think is better.

Checklist:
- [x] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
- [x] Compatibility with Qibo modules (Please edit this section if the current pull request is not compatible with the following branches).
    - [x] Qibo: `master`
    - [x] Qibolab: `main`
    - [x] Qibolab_platforms_qrc: `main`
closed 2024-03-12 by andrea-pasquale