qiboteam/qibocal #758: Rabi amplitude error-handling

rodolfocarobene, 2024-03-20:
Hey, I believe that there is a "bug" in rabi_amplitude. Basically if the fit fails than an error is raised (not the standard warning).
This is because the `durations` variable is uninitialized (if curve_fit fails). I believe we should try to initialize it before (?)

Or at least I think that this is the problem. Please tell me if this is a driver problem (@andrea-pasquale)

I leave below the full error

```
[Qibo 0.2.4|INFO|2024-03-20 13:52:11]: Minimal execution time (sweep): 1.0448896
[Qibocal 0.0.8|INFO|2024-03-20 13:52:12]: Finished acquisition in 1.18 seconds
/home/rodolfo/Venv/qibo/lib/python3.11/site-packages/scipy/optimize/_minpack_py.py:94
5: RuntimeWarning: divide by zero encountered in divide
  transform = 1.0 / sigma
[Qibocal 0.0.8|WARNING|2024-03-20 13:52:12]: Rabi fit failed for qubit 0 due to Resid
uals are not finite in the initial point..
Traceback (most recent call last):
  File "/home/rodolfo/Venv/qibo/bin/qq", line 8, in <module>
    sys.exit(command())
             ^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/lib/python3.11/site-packages/click/core.py", line 115
7, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/lib/python3.11/site-packages/click/core.py", line 107
8, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/lib/python3.11/site-packages/click/core.py", line 168
8, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/lib/python3.11/site-packages/click/core.py", line 143
4, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/lib/python3.11/site-packages/click/core.py", line 783
, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/qibocal/src/qibocal/cli/_base.py", line 72, in auto
    autocalibrate(runcard, folder, force, update)
  File "/home/rodolfo/Venv/qibo/qibocal/src/qibocal/cli/autocalibration.py", line 56,
 in autocalibrate
    for _ in executor.run(mode=ExecutionMode.autocalibration):
  File "/home/rodolfo/Venv/qibo/qibocal/src/qibocal/auto/execute.py", line 142, in ru
n
    completed = task.run(
                ^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/qibocal/src/qibocal/auto/task.py", line 135, in run
    completed.results, completed.results_time = operation.fit(completed.data)
                                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/qibocal/src/qibocal/auto/operation.py", line 41, in w
rapper
    out = func(*args, **kwds)
          ^^^^^^^^^^^^^^^^^^^
  File "/home/rodolfo/Venv/qibo/qibocal/src/qibocal/protocols/characterization/rabi/a
mplitude.py", line 171, in _fit
    return RabiAmplitudeResults(pi_pulse_amplitudes, durations, fitted_parameters, ch
i2)
                                                     ^^^^^^^^^
UnboundLocalError: cannot access local variable 'durations' where it is not associate
d with a value
```

andrea-pasquale, 2024-03-20:
Yes, you are right. Let me open a PR to fix it. Thanks for spotting the issue!

rodolfocarobene, 2024-03-20:
Thanks a lot!

I also believe this (not-correlated) line is outdated:
https://github.com/qiboteam/qibocal/blob/c171ad9cb6e95a37581b636108d5445d194b7a6f/runcards/actions_qq.yml#L121

andrea-pasquale, 2024-03-20:
> Thanks a lot!
> 
> I also believe this (not-correlated) line is outdated:
> 
> https://github.com/qiboteam/qibocal/blob/c171ad9cb6e95a37581b636108d5445d194b7a6f/runcards/actions_qq.yml#L121

Yep, you can now specify the detuning direclty in Hz through a `detuning` key.

PR #759 (2024-03-29): Fixing rabi issue when fitting fails 
Closes #758

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
- [x] Compatibility with Qibo modules (Please edit this section if the current pull request is not compatible with the following branches).
    - [x] Qibo: `master`
    - [x] Qibolab: `main`
    - [x] Qibolab_platforms_qrc: `main`
closed 2024-03-29 by Edoardo-Pedicillo