qiboteam/qibocal #766: Sweeper order for flux protocols

andrea-pasquale, 2024-03-25:


hay-k, 2024-03-25:
See also https://github.com/qiboteam/qibolab/pull/852

RoyStegeman, 2026-11-02:
The title is not informative enough and the referenced PR has been merged, so I guess this can be closed

closed 2026-11-02 by RoyStegeman

alecandido, 2026-11-02:
> The title is not informative enough and the referenced PR has been merged, so I guess this can be closed

Just as historical memory, this problem was limited to the Zurich Instruments driver available at that time. Since some sweepers were available on board, and others were turned into software loops, it was rearranging them to have the maximal number of on-board sweepers, since they could not be placed outer to `for` loops.

If ever we'll have to do this sort of compilation for some sweeper, we should reconsider this issue. Luckily, our present main drivers can sweep everything on hardware. However, in case, I would be inclined to do exactly as the title says: do not apply any hidden optimization, to keep clear guarantees. At most, we should give the option at protocol level for a reversed (or, more in general, permutated) sweeper order, such that the user can explicitly decide how to place. Otherwise, there may be systematic features and so on that are not properly taken into account, if the assumption is that they are executed in a certain order. Or even just for the user watching at the result it may become harder to follow the flow and understand the consequences.