qiboteam/qibocal #790: Dangling references to `n_osc` argument of the ramsey protocol

ckutlu, 2024-04-05:
https://github.com/qiboteam/qibocal/commit/1e47d2f2f1dcfb17f4a14abd359a403623dae5ff replaces the `n_osc` parametrization for the detuned ramsey while the runcards under `qibocal/runcards` still contain references to it. This breaks `qq  auto runcards/routines_benchmarks.yml -o test.out` with the following traceback:

```
Traceback (most recent call last):
  File "/home/caglark/Workspace/QiboBench/qibo.venv/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/home/caglark/Workspace/QiboBench/qibo.venv/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/caglark/Workspace/QiboBench/qibo.venv/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/caglark/Workspace/QiboBench/qibo.venv/lib/python3.9/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/caglark/Workspace/QiboBench/qibo.venv/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/caglark/Workspace/QiboBench/qibo.venv/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/caglark/Projects/qibo/qibocal/src/qibocal/cli/_base.py", line 72, in auto
    autocalibrate(runcard, folder, force, update)
  File "/home/caglark/Projects/qibo/qibocal/src/qibocal/cli/autocalibration.py", line 56, in autocalibrate
    for _ in executor.run(mode=ExecutionMode.autocalibration):
  File "/home/caglark/Projects/qibo/qibocal/src/qibocal/auto/execute.py", line 69, in run
    completed = task.run(
  File "/home/caglark/Projects/qibo/qibocal/src/qibocal/auto/task.py", line 86, in run
    if self.parameters.nshots is None:
  File "/home/caglark/Projects/qibo/qibocal/src/qibocal/auto/task.py", line 59, in parameters
    return self.operation.parameters_type.load(self.action.parameters)
  File "/home/caglark/Projects/qibo/qibocal/src/qibocal/auto/operation.py", line 94, in load
    instantiated_class = cls(**parameters)
TypeError: __init__() got an unexpected keyword argument 'n_osc'
```

andrea-pasquale, 2024-04-05:
Thanks @ckutlu for spotting this error.
Indeed the runcard is outdated, I don't believe that we performed the benchmark again following https://github.com/qiboteam/qibocal/commit/1e47d2f2f1dcfb17f4a14abd359a403623dae5ff. Correct me if I am wrong @stavros11 @Jacfomg @rodolfocarobene 
For the time being I can convert `n_osc` to the corresponding detuning which should be `detuning: 333_333`.

closed 2024-04-05 by scarrazza

stavros11, 2024-04-05:
> Indeed the runcard is outdated, I don't believe that we performed the benchmark again following [1e47d2f](https://github.com/qiboteam/qibocal/commit/1e47d2f2f1dcfb17f4a14abd359a403623dae5ff). Correct me if I am wrong @stavros11 @Jacfomg @rodolfocarobene 

I also don't think we have updated the benchmark to use the frequency detuning. I would expect this to be equivalent to the standard (non detuned) Ramsey in terms of performance, as the only difference is the frequencies of the drive pulses, which should not affect execution time. However, I believe we did not include the non-detuned Ramsey in the original benchmarks.

PR #791 (2024-04-05): Fix outdated runcard
Closes #790 

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ] Tests are passing.
- [ ] Coverage does not decrease.
- [ ] Documentation is updated.
- [ ] Compatibility with Qibo modules (Please edit this section if the current pull request is not compatible with the following branches).
    - [ ] Qibo: `master`
    - [ ] Qibolab: `main`
    - [ ] Qibolab_platforms_qrc: `main`