qiboteam/qibocal #814: allxy_drag_pulse_tuning removed from init

DavidSarlle, 2024-05-02:
allxy_drag_pulse_tuning is not anymore present in the init, so is not available for executing it. But the code is still there.

andrea-pasquale, 2024-05-02:
Thanks @DavidSarlle for opening the issue.
I forgot to remove the file, my idea is that since now we have a drag calibration that is working properly https://github.com/qiboteam/qibocal/blob/main/src/qibocal/protocols/characterization/drag.py we don't need anymore the `allxy_drag_pulse_tuning` protocol.
You should run allxy, drag calibration and allxy again to assess the performance of drag.
Here is an example http://login.qrccluster.com:9000/Qz7dIsHQRjSEs8agnS9mjw==/
Let me know if you are ok with or if we should keep also the  `allxy_drag_pulse_tuning` protocol.

DavidSarlle, 2024-05-02:
Thanks @andrea-pasquale. I have seen the drag routine. I have been using it this morning. It was just as a remember, that the code was still there for allXY_drag_pulse_tuning. I understood that you do not wnated "duplecate" code ;)

But to me the  allxy_drag_pulse_tuning is useful, because you are able to see how the different values of the drag beta parameter affects to your gates. The drag protocol and fit works properly, but some times the value does not improve (or even has a worst performance, at least from the tests that I have being making this morning, probably because the data use to run the protocol or the precission), and is useful to see how the beta parameter really affects the gates.

Having the code, I think is another tool that we offer to the users. But, up to you, you decide :)

andrea-pasquale, 2024-05-02:
> The drag protocol and fit works properly, but some times the value does not improve (or even has a worst performance, at least from the tests that I have being making this morning, probably because [...]

Drag works only if the allxy is already kinda good, it would only improve the "middle" points of the allxy.
I thought also that nobody was using it this is why I planned to remove it. If you find it useful we can keep it.
I will open a PR to restore it properly.

DavidSarlle, 2024-05-02:
thanks @andrea-pasquale . As you want. Keep it or removing it is ok for me. But if we alredy have the code... Is not urgent.

PR #819 (2024-05-07): fix: Restore allxy_drag_tuning
Closes #814 

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ] Tests are passing.
- [ ] Coverage does not decrease.
- [ ] Documentation is updated.
- [ ] Compatibility with Qibo modules (Please edit this section if the current pull request is not compatible with the following branches).
    - [ ] Qibo: `master`
    - [ ] Qibolab: `main`
    - [ ] Qibolab_platforms_qrc: `main`
closed 2024-05-07 by andrea-pasquale