qiboteam/qibocal #817: Utility for automatically scaling `magnitude` to expectation values

marekgluza, 2024-05-03:
When we execute a `qibolab.Pulse` and retrieve the readout magnitude
`platform.execute_pulse_sequence(ps, opts)[q7_ro.serial].magnitude`
we get
[image]

Proposed feature:
- [ ] qibocal after calibrating an $R_X$ $\pi$-pulse can take the maximum and minimum of the Rabi traces and set this as the magnitude level of $\ket 0 \rangle$ and $\ket 1\rangle$ states.
- [ ] these min-max magnitudes can be stored (in the json?)
- [ ] additional utility function which would replace `magnitude` for output visualization
 `platform.execute_pulse_sequence(ps, opts)[q7_ro.serial].minmax_cal_Z_expectation_value`
which would normalize the magnitudes of `q7` and output data $\in [-1,1]$

Expected output:
- if we have many qubits, each readout magnitude will be different so based on the Rabi calibration we can plot the single qubit $\langle Z_k(t)\rangle$ expectation values on top of each other.

Potential use example:
Last week we have run cross-resonance driving on q1,q2,q3 and the script provides output data for magnitudes which are stored and we get
[image]

The problem: Without autocalibration storing the output of a dedicated run of Rabi oscillation for each of the involved qubits then we have no estimate whether the minima of these readout traces of coupled qubits are hitting their respective $|1\rangle$ value or not. In the coupled case the evolution of initial $|0,0,0\rangle$ might not reach $\langle Z_k(t)\rangle =-1$ but just from this plot we don't know what was he minimal expectation value.  (The data in the plot is useless if we don't have access to the min-max of each of the qubits because without min-max we don't know how to stretch and move the data for each qubit; should be automated anyway.)

---
@scarrazza can you add me and @khanhuyengiang to the qibocal team?

andrea-pasquale, 2024-05-06:
Thanks @marekgluza for opening the issue.
Our approach in order to go from voltages to probabilities is to perform a single shot classification experiment, which consists in generating a sequence of 0s and a sequence of 1s and then perform a fit in the IQ plane to classify 0s and 1s.
Would it be possible in your case to adopt the same strategy?

khanhuyengiang, 2024-05-17:
Thank you, @andrea-pasquale, for your response. We currently use a similar strategy for calibrating single-shot experiments. Our proposed feature addresses the arbitrary magnitude obtained using the `SEQUENTIAL` averaging method, where the Rabi oscillations already provide a clear sine curve from which we can infer expectation values of $1$ and $-1$.

This feature will:

1. In `SEQUENTIAL` averaging mode, sweep the arbitrary magnitude of the readout for $\ket{0}$ and $\ket{1}$ for each qubit over time (alternatively, find the min and max of a Rabi pulse on each qubit).
2. Store this magnitude as the baseline.
3. Use this baseline to calibrate all `SEQUENTIAL` readouts, standardizing them from $-1$ to $1$ instead of a raw range (e.g., from 56000 to 62000).

Implementing this feature would be straightforward and intuitive. It will be particularly useful when running multiple qubits using `SEQUENTIAL` readout, making it easier to interpret qubit behavior from the processed output.

For example, the 3-qubit CR plots that Marek mentioned might show Q3 oscillating from $\ket{0}$ to $\ket{1}$, or from $\ket{0}$ to $\ket{+}$, but this is not evident from the raw measurement values:

>![three_CR_010_201](https://private-user-images.githubusercontent.com/6939096/327618286-6f490360-eadb-422e-bb7d-612b700abe1c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU5MjQ5ODIsIm5iZiI6MTcxNTkyNDY4MiwicGF0aCI6Ii82OTM5MDk2LzMyNzYxODI4Ni02ZjQ5MDM2MC1lYWRiLTQyMmUtYmI3ZC02MTJiNzAwYWJlMWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MTdUMDU0NDQyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTA2ZWU2NDk1ODBmYTE4YTljNTUzOTJmODI3ZGRmOTAyNzIyNDM3NzExODc4YzBiMDdhYzQ4NDZmYTM2OWRlZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.dgBPTijMPWsbebBG3uur_LeOmFAcJZrl6ihOsaI-FG4)

I have make a proof of concept code at [this new branch](https://github.com/qiboteam/qibocal/tree/sequential_mode_expectation_calibration/src/qibocal/protocols/sequential_mode_expectation_calibration/proof_of_concept), that can be run and tested on our hardware at NTU. Would it be reasonable to add this to qibocal?

andrea-pasquale, 2024-05-18:
Thanks @khanhuyengiang for the reply.
I think that this is the behavior intended for `magnitude` given that there is no post-processing applied.
I had a quick look at the code and I understand the idea. It is not the standard procedure but I believe that it could work.
Regarding this normalization, in qibocal we also provide a Rabi protocol where the output is already normalized https://github.com/qiboteam/qibocal/blob/main/src/qibocal/protocols/rabi/amplitude.py. We can do this by retrieving samples from the instrument:
https://github.com/qiboteam/qibocal/blob/4c23ced7d5ee1e93d758e75179308a9d13c6da90/src/qibocal/protocols/rabi/amplitude.py#L94-L114
Could this work for you? I think that the outcome should be pretty much the same :)

closed 2025-02-14 by andrea-pasquale