qiboteam/qibocal #821: Cross Resonance Routine

sorewachigauyo, 2024-05-03:
To implement a basic routine for the calibration of cross-resonance (CR) gates.

This involves sending a pulse on the control qubit with the frequency of the target qubit. We vary the pulse length to observe a Rabi oscillation. Based on the state of the control qubit, there will be a splitting of the Rabi frequencies, and we can select a pulse length $t$ such that the phase difference between the two oscillations is $\pi$. This then implements a CNOT gate.

[image]

A basic CR calibration is as follows:
```python

sweep = np.arange(0, 1000, 20)
res1 = np.zeros(len(sweep))
res2 = np.zeros(len(sweep))

crtl_pi_pulse = platform.create_RX_pulse(qubit=CRTL_QUBIT, start=5)
tgt_ro_pulse = platform.create_qubit_readout_pulse(qubit=TGT_QUBIT, start=crtl_pi_pulse.finish)
cr_pulse = platform.create_RX_pulse(qubit=TGT_QUBIT, start=crtl_pi_pulse.finish)
cr_pulse.channel = crtl_pi_pulse.channel
cr_pulse.amplitude = 1

# We first perform a CR sweep when the control qubit is at the |0> state.
# This is equivalent to a typical Rabi length experiment
ps = PulseSequence(*[cr_pulse, tgt_ro_pulse])
for idx, t in enumerate(sweep):
    cr_pulse.duration = t
    tgt_ro_pulse.start = cr_pulse.finish
    res1[idx] = platform.execute_pulse_sequence(ps, opts)[tgt_ro_pulse.serial].magnitude

# Next, we add a pi-pulse on the control qubit at the start to get results for the |1> state.
ps.add(crtl_pi_pulse)
for idx, t in enumerate(sweep):
    cr_pulse.duration = t
    tgt_ro_pulse.start = cr_pulse.finish
    res2[idx] = platform.execute_pulse_sequence(ps, opts)[tgt_ro_pulse.serial].magnitude
```

The frequency of each Rabi oscillation is proportional to the amplitude of `cr_pulse`, so to obtain an optimal parameter for the amplitude, a 2D chevron of CR pulse amplitude and length can be performed.
``` python
sweeper_amplitude = Sweeper(
    Parameter.amplitude,
    amplitude_range,
    pulses=[cr_pulse],
    type=SweeperType.FACTOR,
)
sweeper_duration = Sweeper(
    Parameter.duration,
    duration_range,
    pulses=[cr_pulse],
    type=SweeperType.ABSOLUTE,
)
results = platform.sweep(
    ps,
    opts,
    sweeper_duration,
    sweeper_amplitude,
)
# and repeat for the other state of the control qubit
```

By also simultaneously driving the target qubit on resonance, an approximate interaction forms
$A_{CR} (\omega_{ZX} ZX + \omega_{IX} IX) + A_t \omega_{R2} IX $
where $A_{CR}$ is the amplitude of `cr_pulse`, $\omega_{ZX}, \omega_{IX}$ is determined by the target qubit Rabi frequency $\omega_{R2}$, effective qubit-qubit coupling, detunings and anharmonicites and $A_t$ is the amplitude of the second drive. 

A basic experiment with the second drive is as follows:

```python
crtl_pi_pulse = platform.create_RX_pulse(qubit=CRTL_QUBIT, start=5)
tgt_ro_pulse = platform.create_qubit_readout_pulse(qubit=TGT_QUBIT, start=crtl_pi_pulse.finish)
cr_pulse = platform.create_RX_pulse(qubit=TGT_QUBIT, start=crtl_pi_pulse.finish)
#cr_pulse.relative_phase = np.pi
IX_pulse = platform.create_RX_pulse(qubit=TGT_QUBIT, start=crtl_pi_pulse.finish)
IX_pulse.amplitude = 0.4
cr_pulse.channel = crtl_pi_pulse.channel
cr_pulse.amplitude = 1

ps = PulseSequence(*[cr_pulse, tgt_ro_pulse, IX_pulse])
sweeper_duration = Sweeper(
    Parameter.duration,
    duration_range,
    pulses=[cr_pulse, IX_pulse],
    type=SweeperType.ABSOLUTE,
)
results_crtl_gnd = platform.sweep(
    ps,
    opts,
    sweeper_duration
)
ps.add(crtl_pi_pulse)
results_crtl_exc = platform.sweep(
    ps,
    opts,
    sweeper_duration
)
```

aorgazf, 2024-05-15:
Thanks @sorewachigauyo for raising this. It is also needed for the upcoming batch of TII's 2q chips.

sorewachigauyo, 2024-09-17:
Hi @jevillegasd,

If its not too sensitive, could you give some more information about the qubit system you are using? Given that the qubits are 80MHz apart, it is strange that you do not see an effect or even crosstalk-driven Rabi oscillations.

We can take it to emails if needed.

jevillegasd, 2024-09-18:
Absolutly the QPU is a three qubit chip with fixed frequency qubits coupled together in a straight line (you can see the platform here https://github.com/qiboteam/qibolab_platforms_qrc/commit/77b7b3236ad393f657c7f4ac85b69df9ccea92ca). The chip has an additional witness qubit and a resonator. Here I list the most relevant parameters.

  | f_ro hp , GHz | f_ro lp , GHz | $\chi$ , MHz | f_01 , GHz | f_02/2 , GHz | T1 [us] | T2 [us]
-- | -- | -- | -- | -- | -- | -- | --
Q1 | 6.9111 | 6.9060 | 5.03 | 5.312 | 5.150 | 6.3 | 7.1
Q2 | 7.020138 | 7.015958 | 4.17 | 5.393 | 5.242 | 4.5 | 3.6
Q3 | 7.140432 | 7.135537 | 4.89 | 5.913 | 5.764 | 5.2 | 5.4
QW | 7.626598 | 7.621588 | 5.01 | 4.908 | 4.768 |   |  
Free Res | 7.892598 | 7.892598 | 0 |   |   |   |

jevillegasd, 2024-09-18:
I've run the routine using [0,0] as target and control qubits, which should just make a normal Rabi, this to confirm if its working. And here is what I get

[image]
which is arguably good.

I will try increasing the power and see where that takes us.

Edoardo-Pedicillo, 2024-09-18:
> I will try increasing the power and see where that takes us.

Have you tried to run a cross resonance chevron ?

sorewachigauyo, 2024-09-18:
The qubits are coupled directly or by bus resonator? Do you have an estimate of the coupling strength from designed values or measured by a two-qubit simultaneous on-resonance Ramsey?

jevillegasd, 2024-09-18:
> > I will try increasing the power and see where that takes us.
> 
> Have you tried to run a cross resonance chevron ?
I still need to make the version with sequences for that as well so that I can use it with QBlox.

jevillegasd, 2024-09-18:
> The qubits are coupled directly or by bus resonator? Do you have an estimate of the coupling strength from designed values or measured by a two-qubit simultaneous on-resonance Ramsey?

They are directly coupled with a transmission line with a coupling capacitance of 0.5 fF at each transmon, this is about 10 MHz by design. We have not tried to extract this value using simultaneous Ramsey tests, could you explain how can we get this?

andrea-pasquale, 2024-09-18:
> They are directly coupled with a transmission line with a coupling capacitance of 0.5 fF at each transmon, this is about 10 MHz by design. We have not tried to extract this value using simultaneous [...]

If @sorewachigauyo is referring to the ZZ coupling, there is a PR where we have a draft for that protocol #970.
Essentially if you do a Ramsey experiment on one qubit will keeping another qubit in state 1, if you compare it with the original ramsey from the phase shift in the oscillations you should be able to get the ZZ coupling.
However, we are using again duration sweepers so it will not work out of the box with qblox.

sorewachigauyo, 2024-09-18:
Yeah, #970 should give you the coupling strength for this direct coupling and checks if the qubits are coupled or not. Based on the coherence times you measured, you should observe the Ramsey oscillations within the coherence.