qiboteam/qibocal #835: New experiments

rodolfocarobene, 2024-05-08:
Hi, I know that there already is #408 regarding this, but I prefer to have an issue that I can close "fast".

At UNIMIB, we started using qibocal and we would like to add some new experiments that are standard calibration ones (usually done at the beginning of the characterization phase). In particular these are:

```[tasklist]
### Tasks
- [ ] Qubit power spectroscopy (like punchout, but for qubit)
- [ ] Ramsey changing the frequency (it should be easier to see the correct pi pulse frequency, independently from the fit)
- [ ] Rabi changing the frequency (it should be possible to also tune the frequency with this, leading to an overall easier calibration) (#608)
```

Next week I will have some time to open the specific PR. All the experiments seems fairly easy / similar to ones w already have so it should not take much time. I will need some help with the testing phase (on hardware) probably, but we will see

andrea-pasquale, 2024-06-05:
@rodolfocarobene regarding the experiment with ramsey changing frequency what do you mean exactly? Some sort of 2D plots or just the possibility of changing direclty the drive frequency as input parameter? The second one is already implemented :)

rodolfocarobene, 2024-06-05:
I meant a 2D plot, at NIST they use it since the fit (exponential + oscillation) can sometimes be difficult. However, I never saw it and I'm not sure of what should we expect if we try it

closed 2024-06-25 by rodolfocarobene