qiboteam/qibocal #850: Qblox mixer calibration routine

aorgazf, 2024-05-15:
Similar to what it is being performed with QM, and once https://github.com/qiboteam/qibolab/issues/900 is completed, a routine would be necessary to calibrate mixers using resonators and qubits' responses.
This routine is quite important when attempting to calibrate QPUs with many qubits.

hay-k, 2024-05-15:
@aorgazf I believe you were already experimenting with mixer calibration a few weeks back, by directly manipulating the said instrument settings. If you have any insight, suggestion, draft implementation, reference(s) to paper(s) that are useful for understanding how to best implement this routine, please link them here.

andrea-pasquale, 2025-05-30:
Closing in favor of #1164

closed 2025-05-31 by alecandido