qiboteam/qibocal #883: Fano functions

rodolfocarobene, 2024-06-17:
Hi! 
We would like to try to implement the Fano fit for the resonance fit. This should help to better fit the asymmetric resonance and also provide a direct estimation of the internal quality factor Q_i and of Q_c.
I think @marcogobbo could have a try at this :-)

Is it ok for you?

Edoardo-Pedicillo, 2024-06-19:
Hi @rodolfocarobene, 
Sure, go on. 
In the past, we tried the fit described in https://arxiv.org/pdf/1410.3365, it was unstable, but maybe by fitting the whole complex signal, we could extract more information.

sorewachigauyo, 2024-07-10:
There is also https://github.com/sebastianprobst/resonator_tools/tree/master which is the implementation of the paper, not sure if you have already examined it.

alecandido, 2024-07-10:
Thanks @sorewachigauyo, I was not aware of it.

However, I took a look at the code, and it doesn't seem much better than what we have in #917 (a lot of code, unformatted, undocumented, and more or less repeating the exact same steps [concerning the circle fit](https://github.com/sebastianprobst/resonator_tools/blob/bbd6bca255eb7ebdf550fde9f6c526a7d0791eac/resonator_tools/circlefit.py#L115-L171)).

It will be definitely useful to check whether the final product of #917 achieves feature parity with that repo, but I would not attempt copying that code, now that we already have a fit, working at least for some data (just because I believe to be a less efficient strategy, not for any other reason).

marcogobbo, 2024-07-10:
Hi @sorewachigauyo!

Yes, in the various tests I have done, I also tested the repository you mentioned. Actually, that code was then carried forward by QKIT from which I took part for the PR!

I didn't like some of their choices, like the complete fit for the cable delay or the fit of the circle.

The references that I am following are:

- [Efficient and robust analysis of complex scattering data under noise in microwave resonators](https://doi.org/10.1063/1.4907935) by S. Probst et al.
- Jiansong Gao's [thesis](https://thesis.library.caltech.edu/2530/1/thesismain_0610.pdf) (Appendix E)

sorewachigauyo, 2024-07-11:
There is also https://github.com/steelelab-delft/stlab/blob/master/misc/S11fit.py which my group is using lately for noisy data

closed 2024-08-13 by andrea-pasquale