qiboteam/qibocal #891: Timing and sweepers

alecandido, 2024-06-19:
This issue apparently involves all the routines with a time-related sweeper, and it spans Qibolab as well.

It was initially spotted in https://github.com/qiboteam/qibolab/pull/914#discussion_r1645439749 that the start time of the readout pulse in a Rabi experiment should be played right after the drive pulse is finished, for all the values in the sweepers.

However, the only way the Qibocal routine can specify this, is by playing the readout pulse at the end of the drive for the **initial** value of the sweeper. Hoping for the best.

This is not a full specification of how the experiment should be performed, since the start value of the readout should be updated by the sweeper, but there is no specification of that in the `platform.sweep()` inputs.
Instead, the acceptable solutions would be:
- sweep the start of the readout together with the duration of the drive (requires simultaneous sweepers https://github.com/qiboteam/qibolab/issues/692)
- sweep the duration of a readout delay together with the duration of the drive (it could be done with a single sweeper, but it would require a delay https://github.com/qiboteam/qibolab/issues/682)
- add an `align()` instruction between the two pulss (it would require the align itself, https://github.com/qiboteam/qibolab/issues/745)

Current Qibocal is more or less the best it can be done with the present Qibolab, but still the implementation is flaky or just wrong, and we should upgrade it as soon as Qibolab will support one of the options above.

alecandido, 2024-06-19:
I still have to confirm, but I suspect that this is not an issue with most of the drivers, because of an implicit push for the measurement at the end of the experiment, that would be an issue on its own (because we lose part of the control over the readout pulses timing).

Is this the case? @stavros11 @hay-k

andrea-pasquale, 2024-10-01:
@alecandido can we close this? It feels that this was already solved by qibolab 0.2

alecandido, 2024-10-01:
@andrea-pasquale you're definitely right, this is actually solved in Qibolab 0.2. All the options above are available in current Qibolab.

I believe @stavros11 already updated Rabi in #990, with and without using `Align` instructions.
If he confirms, we can definitely close the issue.

stavros11, 2024-10-31:
Sorry for the late reply on this, I just found it while going through old marked notifications.

Indeed, duration sweepers are much more transparent in qibolab 0.2. To add to the last comment, now multiple routines involving such sweepers have been implemented in #990, including Rabi length, T1, Ramsey and spin echo. These are mainly taking advantage of parallel sweeps and have been tested on QM. If you agree with these implementations feel free to close the issue.

Also note that `pulse.start` has been dropped in 0.2, so the only time related parameter that can be swept is `duration`, either of a `Pulse` or a `Delay`, which also simplifies things a bit. The only open problem is to make sure uniform behavior across instruments, however this is qibolab's problem and we can only test it after we have more drivers in 0.2.

closed 2025-01-09 by andrea-pasquale