qiboteam/qibocal #895: Fine tuning of CZ flux pulses.

aorgazf, 2024-06-20:
Hi,
The results of the `chevron` fit provide a rough estimate of the flux pulse that implements a CZ gate. With that, one runs `cz_virtualz` multiple times, varying the amplitude of the flux pulse, until the gate effectively implements a rotation of pi. This process is quite slow.
It would be nice to have a routine that automatically sweeps around the duration and amplitude parameters found by the chevron routine and calculates the rotation between the qubits, returning the point closest to pi.

andrea-pasquale, 2024-06-21:
Hi @aorgazf, something along those lines was originally performed in #823.
The code has not been tested since the last warm-up.

andrea-pasquale, 2024-08-12:
Closing this since #823 got merged

closed 2024-08-12 by andrea-pasquale