qiboteam/qibocal #897: Rabi fit failing in some instances

andrea-pasquale, 2024-06-21:
While testing the emulator I saw that the rabi fit tends to fail even if the output is quite well defined.

[image]

I think that the error is related to the fft guess, we need to investigate this more.

alecandido, 2024-06-21:
Here we can see multiple oscillations, and the emulator is returning a fully imbalanced signal, since it's always positive (at the end of the day, it is always a probability, despite working as a signal).

Thus, we have to subtract from the FFT the 0th moment, i.e. the integral itself, otherwise it will dominate the transformed function, and it may lead to a guess for initial parameters that is very far from the desired global minimum.

Most likely, this was already done in a former version of Qibocal, but it got lost in the current one.

sorewachigauyo, 2024-06-24:
On this topic, it might be nice to add tests for individual routines using the emulator (single qubit for now), since the results are deterministic, but that might increase the CI/CD cost.

andrea-pasquale, 2024-06-24:
Thanks @sorewachigauyo, indeed we could consider doing something like this.
After discussing with @alecandido, a possible way to speed up the execution time with the emulator could be to run using `AveragingMode.SEQUENTIAL`. We just need to come up with a mechanism to plug this information into tests.

alecandido, 2024-06-24:
To be more precise: that's a possible way, provided that the emulator is instructed to use it properly (which I believe is not, right now).

`AveragingMode.SEQUENTIAL` makes the shots the innermost loop. If this information is taken into account, since the evolution is deterministic, the emulator should not re-evolve for each and every shot, just sampling the final state as many times as required.
This would make the shots almost priceless (i.e. for any reasonable amount of shots, the price should be pretty cheap, in comparison to the evolution).
But you will pay for each and every parameters' combination.

In any case, I fully support using the emulator for testing the routines in the CI, as many as possible (even when the results are not fully meaningful - but it will always be incredibly better than with dummy). But I would dedicate an issue on its own ^^

closed 2024-07-10 by andrea-pasquale