qiboteam/qibocal #920: Handle some exceptions gracefully

alecandido, 2024-07-05:
There are a number of errors that are related to the interface, in which we'd like to just communicate something to the user.

E.g. when a directory is already existing, and about being overwritten, but no force has been set.

What we are doing is just to let Python fail through its exception mechanism, and throw a potentially lengthy stack trace in user's face.
This was the best possible choice in the short term, because these stack traces are as ugly as they are helpful, since they allow reconstructing the calling path.

However, for those errors that are largely expected (like the one above), we could come with a better interface, avoiding the stack trace. Possibly even in scripts.

This is just a matter of UI, and it should not affect in any way the internal workflow of the library.
The most external elements handling execution should receive the errors as they are, but filter them right before passing to the user. They can be recognized by having suitable custom exceptions, all subclassing from a single `UIException`, or something like that.

merelyacryptid, 2026-09-30:
hi! @alecandido  i'd like to take this issue, my changes would be

-Add a UIException base class for expected, user-facing errors
-Catch it at the CLI entry point, print a clean message instead of the traceback, exit non-zero
-convert the directory-exists-without---force case as the first concrete example
-leave other candidate cases for follow-up PRs rather than trying to convert everything at once

Let me know if that split (mechanism + one converted example) matches what you had in mind, or if you want more cases bundled into the first PR.

alecandido, 2026-09-30:
Hi @merelyacryptid: I labeled this as "good first issue", but in the hindsight, it is actual a bit harder, since it is spread around the whole project. Possibly, I also had in mind some specific collaborator on-training, but I don't even remember.

I would not really recommend this to a first time contributor. And if you are interested in just contributing, I would recommend you to contact me in private (my addresses are public, choose the one most suited to you), and we can discuss.

If you are specifically interested in this issue, then keep going. If the proposal is interesting and useful, we will certainly consider it.

But first things first, let me remove the label...