qiboteam/qibocal #926: Alleviate or document environment dependence

alecandido, 2024-07-09:
After #869 there will be a default executor, already instantiated to be ready to be used.

https://github.com/qiboteam/qibocal/blob/d63bc73efd189cf316c7bd31bde6341af6151a3c/src/qibocal/__init__.py#L10

The best would be to only instantiate it if needed. But this may not be completely possible, in order to retain the `from qibocal.routines import ...` feature.

If present, it will rely on the `QIBO_PLATFORM` environment variable, 
https://github.com/qiboteam/qibocal/blob/d63bc73efd189cf316c7bd31bde6341af6151a3c/src/qibocal/auto/execute.py#L100-L108
thus we should be very careful in documenting that this will interfere with the execution, even when you're not using it.

A more complicate option (but more intuitive for the user) would be to create the executor immediately, but postpone the creation of the platform, e.g. until the first routine call.

andrea-pasquale, 2024-07-09:
I was thinking about this yesterday... Perhaps it would be better to rely on `dummy` by default to avoid issues.
If `QIBO_PLATFORM` is pointing to an outdated platform the code fail.

alecandido, 2024-07-09:
Yes, it is definitely something to take care of. I'm not even sure it should be `dummy`, the solution I like the most is the latest.

But most of the time it will already be `dummy`, if that variable is unset (which maybe it should be even on the cluster), so it is not that urgent...

alecandido, 2025-02-27:
It's a separate issue, but I even wonder whether we need the default executor at all. The context manager (`with Executor.open(...) as e:`) pattern seems actually good enough on its own...

Dropping it would limit even more the code that executes at import-time.

closed 2025-05-05 by Edoardo-Pedicillo