qiboteam/qibocal #958: `spin_echo` fails to execute. Wait time as `float` not supported in QM

igres26, 2024-08-01:
Error given when trying to execute the `spin_echo` routine using quantum machines control electronics:

```
2024-08-01 10:54:57,352 - qm - INFO     - Sending program to QOP for compilation
2024-08-01 10:54:57,428 - qm - ERROR    - Unexpected error: Failed to compile job
2024-08-01 10:54:57,429 - qm - ERROR    - Time inside wait statement cannot be of type real (@ File "/nfs/users/sergi.ramos/qibolab/src/qibolab/instruments/qm/sequence.py", line 261: qua.wait(qmpulse.wait_cycles, qmpulse.element) )
2024-08-01 10:54:57,429 - qm - ERROR    - Time inside wait statement cannot be of type real (@ File "/nfs/users/sergi.ramos/qibolab/src/qibolab/instruments/qm/sequence.py", line 261: qua.wait(qmpulse.wait_cycles, qmpulse.element) )
2024-08-01 10:54:57,429 - qm - ERROR    - Job 1719486296409 failed. Failed to execute program.
Traceback (most recent call last):
  File "/nfs/users/sergi.ramos/qiboenv/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/cli/_base.py", line 74, in auto
    autocalibrate(runcard, folder, force, update)
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/cli/autocalibration.py", line 42, in autocalibrate
    history = runcard.run(
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/auto/runcard.py", line 63, in run
    instance.run_protocol(
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/auto/execute.py", line 209, in run_protocol
    completed = task.run(platform=self.platform, targets=self.targets, mode=mode)
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/auto/task.py", line 149, in run
    completed.data, completed.data_time = operation.acquisition(
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/auto/operation.py", line 42, in wrapper
    out = func(*args, **kwds)
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/protocols/coherence/spin_echo.py", line 101, in _acquisition
    results = [
  File "/nfs/users/sergi.ramos/qibocal/src/qibocal/protocols/coherence/spin_echo.py", line 102, in <listcomp>
    platform.execute_pulse_sequence(sequence, options) for sequence in sequences
  File "/nfs/users/sergi.ramos/qibolab/src/qibolab/platform/platform.py", line 196, in execute_pulse_sequence
    return self._execute(sequence, options, **kwargs)
  File "/nfs/users/sergi.ramos/qibolab/src/qibolab/platform/platform.py", line 170, in _execute
    new_result = instrument.play(
  File "/nfs/users/sergi.ramos/qibolab/src/qibolab/instruments/qm/controller.py", line 322, in play
    return self.sweep(qubits, couplers, sequence, options)
  File "/nfs/users/sergi.ramos/qibolab/src/qibolab/instruments/qm/controller.py", line 369, in sweep
    result = self.execute_program(experiment)
  File "/nfs/users/sergi.ramos/qibolab/src/qibolab/instruments/qm/controller.py", line 256, in execute_program
    return machine.execute(program)
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/qm/QuantumMachine.py", line 236, in execute
    pending_job = self._queue.add(program, standardized_compiler_options)
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/qm/jobs/job_queue.py", line 77, in add
    job = self._insert(program, InsertDirection.end, compiler_options)
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/qm/jobs/job_queue.py", line 145, in _insert
    job_id = self._frontend.add_to_queue(
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/qm/api/base_api.py", line 28, in wrapped
    return func(*args, **kwargs)
  File "/nfs/users/sergi.ramos/qiboenv/lib/python3.10/site-packages/qm/api/frontend_api.py", line 217, in add_to_queue
    raise FailedToAddJobToQueueException(f"Job {job_id} failed. Failed to execute program.")
qm.exceptions.FailedToAddJobToQueueException: Job 1719486296409 failed. Failed to execute program.
```

Talking to @andrea-pasquale, seems to be related to the wait times not being `integer` values, and creating a conflict with quantum machines.

andrea-pasquale, 2024-08-01:
This is mostly related to qibolab developers. If float should be supported as pulse duration the fix should be in qibolab, if instead the duration can only be a integer I can fix it here. Thoughts @stavros11 @alecandido @hay-k ?

andrea-pasquale, 2024-08-02:
Thanks @alecandido, shall I do a quick patch here and then convert it to float once we have 0.2?
In this way we can run `spin_echo` on QM.

alecandido, 2024-08-02:
> Thanks @alecandido, shall I do a quick patch here and then convert it to float once we have 0.2?
> In this way we can run `spin_echo` on QM.

Indeed, I'd suggest you ant simply an `int(...)` call where needed, and remove that whenever we'll come to 0.2.

andrea-pasquale, 2024-08-02:
Fixed in https://github.com/qiboteam/qibocal/pull/961/commits/f8e36a085a14ad0496661233db4b021a236b1cea

closed 2024-08-12 by andrea-pasquale