qiboteam/qibocal #966: Update toggle test failing at random

alecandido, 2024-08-02:
I haven't identified yet any random number involved, but it seems that this test is failing at random

https://github.com/qiboteam/qibocal/blob/b0b569a56c6d669f73feb62b6aac5131ce220532/tests/test_task_options.py#L103-L125

https://github.com/qiboteam/qibocal/actions/runs/10219428483/job/28277670887
(the second attempt is passing, without having changed anything - also noticed that is just failing on win-py3.11 combination, but working on win-py3.9, linux, and darwin, so it seems to be just a chance)

alecandido, 2024-08-02:
@andrea-pasquale it seems you wrote this test. Any clue about?

(it should not be difficult to debug, anyone can do it, so it doesn't have to be necessarily you - but maybe you already had an idea)

andrea-pasquale, 2024-08-12:
Yes, I know.
The test is failing when the lorentzian fit fails therefore the parameters are not updated.
As a patch we could keep the test only for the single shot classification which shouldn't ever fail.

alecandido, 2024-08-12:
> The test is failing when the lorentzian fit fails therefore the parameters are not updated.

But how is it possible that the Lorentzian fit will pass just rerunning?
Is `curve_fit` using some sort of random numbers internally? In case: could we fix the seed for that?

andrea-pasquale, 2024-08-12:
The output is different every time because we are testing with `dummy`, right?
Given that we don't fix the seed of the platform several runs will have different results.
So one option is to fix the seed of `dummy`

alecandido, 2024-08-12:
That is definitely a seed that should be fixed.

Then, we may still decide that the fit is too unstable to be tested in the CI, especially with dummy. And we'd rather design a closure test for it.
But all seeds should be fixed, for reproducibility.

RoyStegeman, 2026-10-31:
It seems that Andrea implemented his proposed solution to use single shot to avoid the fit instability of a Lorentzian fit. @alecandido probably this issue should be closed and, if you want to keep tracking it, a new issue opened for seeding the dummy.

closed 2026-10-31 by alecandido