qiboteam/qibocal #1023: Error values of T2

HishamKazim, 2024-10-23:
This issue concerns the large discrepancy in the error values of T2 as shown in the PR below:

https://github.com/qiboteam/qibolab_platforms_qrc/pull/191

The error is greater than the actual T2 value (much greater).

alecandido, 2024-10-23:
In general, having an uncertainty greater than the value is possible, but it means that the value is compatible with 0 (so it's either vanishing, or you don't even know it's sign).

And this is clearly not the case of a fit like the following:
[image]
it is pretty clear that the data are exponentially decaying, and the fit manage to match the data pretty well.
So, you certainly have no doubt on the sign of the exponential (ascending or descending), while the value reported is 30 +/- 40 us.

http://login.qrccluster.com:9000/qYLzOD1ET9yCXMAr0ZsNmA==/#T2

We're not consuming in any way the uncertainty on parameters during the plotting, while the central value is used to produce the fit.
I'd say that the plot of the fit confirms the central value was correctly found. The uncertainty should definitely be investigated.

Edoardo-Pedicillo, 2024-10-24:
I have done a small investigation, the parameters and errors of the fit of the normalized data (`perr` and `popt` in the code ) are 
```
popt = [0.00342815,  1. ,         1.2564083,  1.72139401, 1.70591943]
perr =  [0.37864146,  0.04844122,  0.23573182,  2.62129502,  2.53873094]

```
even though, the fit is good `curve_fit` is not able to estimate the errors properly.

I have tried to remove the sinusoidal dependency from the fit function and remove the data normalization on the 
y axis (probabilities):
```py

   def ramsey_fit(x, offset, amplitude, delta, phase, decay):
       """Dumped sinusoidal fit."""     
       # return offset + amplitude * np.sin(x * delta + phase) * np.exp(-x * decay)
       return offset + amplitude * np.exp(-x * decay)

```
and I have obtained better results 

```
popt = [ 0.46887992  0.47956018 25.69336208  0.          1.71742433]
perr =  [0.00568939 0.00512151 0.         0.         0.04029972]
```
It follow the report in the second case:
[image]

The last value in the `popt` list is the one we use for the evaluation of T2, it differ slightly in the two cases, but the second case should be the most reliable one (according to the physics laws of the system). 
So it is clear that the problem is related to how `curve_fit` evaluates the errors, I have tried to dig deeply but I was not able to find something significant.

alecandido, 2024-10-24:
Ok, I was not aware of the sinusoidal dependency was present even in the fit of T2. I would have thought it was just for the Ramsey detuned.

My potential explanation for what is happening is the following:
- the fit has an arbitrary sine at disposal, with an arbitrary amplitude, frequency and phase (so it is free to be scaled on the vertical on the horizontal axis, and translated horizontally)
- since no sine is actually present, you would expect that to fit to a 0 amplitude
    - however, the 0 amplitude is also achieved by completely ignoring the sine 
- at this point, you have a quite large exponential decay (not even an order of magnitude), and a free sine available
- the exponential could be approximated well-enough by a second order polynomial (a parabola)
- it is possible to fit the intercept, slop, and curvature for the sine in order to match the polynomial (even the magnified sine is well approximated by a parabola)
- since it is actually exponential, even the exponential term in the function can fit the behavior

At this point, you get two different terms that could actually both fit the exponential decay, and you develop a flat(/flatter) direction, which is a combination of the two terms.
Since the error is estimated with the inverse curvature near the minimum, you find a minimum at a random (noise-driven, or at least noise-prone) point in the flat direction, and the curvature is very large.
If the flat direction is neither the sine nor the exponential, but they are sufficiently similar to each other to be a combination of the two (i.e. the eigenvector of the Hessian in the parameter space), the wide curvature projects on both directions (exponential parameters and sine parameters), yielding large errors for both.

Most likely, the best is just freezing the sine while measuring T2. As you did in your experiment.

Edoardo-Pedicillo, 2024-10-25:
Thanks @alecandido for the explanation.

> Ok, I was not aware of the sinusoidal dependency was present even in the fit of T2. I would have thought it was just for the Ramsey detuned.

The sinusoidal formula is used only in the Ramsey experiment(s), indeed what is shown by @HishamKazim is a Ramsey with detuning 0.

> Most likely, the best is just freezing the sine while measuring T2. As you did in your experiment.

@HishamKazim can you try to repeat the T2 evaluation using the `T2` routine (https://qibo.science/qibocal/stable/protocols/t2/t2.html) and upload here the results ?

alecandido, 2024-10-25:
Yes, sorry to have said something incorrect

> Ok, I was not aware of the sinusoidal dependency was present even in the fit of T2. I would have thought it was just for the Ramsey detuned.

The sine is just in Ramsey, the T2 routine @Edoardo-Pedicillo is proposing is actually just the exponential decay. And that's correct

andrea-pasquale, 2024-10-25:
We had similar discussions in the past about this. The reason why we use Rasmey to measure T2 is that you cannot know before running the experiment whether your drive frequency is sufficiently close to the actual qubit frequency to avoid oscillations. In fact, if you fit with just an exponential and there are oscillations the fit is most likely going to fail (we saw this especially for the monitoring...).
At this point we can try to pre-process the data before doing the fit in order to choose whether we should perform an exponential fit or an exponential fit + sinusoidal term. On the top my head I'm thinking that we could do a FFT to detect if there is an oscillating term, if we find it we fit also with a sin if not we just fit with the exponential. Thoughts?

alecandido, 2024-10-25:
What I would suggest is to do that in two steps: run with detuning, even a small one, to measure more precisely which it is (which is what @HishamKazim already did in http://login.qrccluster.com:9000/qYLzOD1ET9yCXMAr0ZsNmA==/#Ramsey%20Detuned), and then run pure T2, to fit the exponential decay.

In principle, you could do that even in a single protocol, but there is no advantage over the composition.
I would try to rerun the same calibration reported above, just using the T2 routine with pure exponential decay. If the oscillations are very mild (because you removed the sine before), you should still be able to fit quite precisely the exponential decay. Even if the chi2 may skyrocket for small errors and non-negligible oscillatory behavior, the minimum should be identified very precisely anyhow, and the fit will not fail.

The direction I see for improvements is just to optimize the strategy to improve the oscillatory behavior as much as possible. I.e. everything that happens before the T2 protocol, and the pure exponential decay fit. But eventually, you should use the T2 protocol to extract the T2 value.

andrea-pasquale, 2024-10-25:
> What I would suggest is to do that in two steps: run with detuning, even a small one, to measure more precisely which it is.

You need to be careful about a small detuning, given to get a valid correction on the qubit frequency you need be in the situation where the detuning is larger compared to the frequency correction. This is why we usually choose to a have a large detuning of a few MHz when we run ramsey.

In general you can always say something like "I will run with detuning and afterwards I will run just the T2 experiment to measure T2". However, in some cases even if you correct the qubit frequency T2 can exhibit a non monotonic behavior due to some spurious qubit-qubit coupling.

This is why I'm suggesting that we should do something within the protocol itself.

HishamKazim, 2024-10-25:
So I reran the ramsey, T2 (using the ramsey with detuning 0), and the T2 runcard as shown in the report below: http://login.qrccluster.com:9000/jX1y6NXDTvmbPHLEu5DIZQ==/. Let me know your thoughts @Edoardo-Pedicillo @alecandido @andrea-pasquale 
[image]
[image]

andrea-pasquale, 2024-10-25:
Thanks @HishamKazim for the plots. Indeed, this confirms that an exponential fit  should be enough.

alecandido, 2024-10-25:
Indeed, this is showing that in this case the exponential fit is doing its job. Not only because of the uncertainty reduction, but also because the central value of the exponential decay shifted more than its error in the pure exponential fit[*].

However, this is just a phenomenological evidence that is working fine for this specific case.

My personal advice would be to keep using consistently this strategy (that in this case worked well enough) and then accumulate evidence for when it's failing, in order to improve consistently the strategy.
Most likely, the current one is not perfect at all. But without an example of how it could fail, it is harder to improve.

---

[*] It is interesting the shift in the central value in the decay, and makes pretty clear the sine contamination. However, to complete the empirical proof (i.e. attribute the responsibility for this specific case), we should actually fit the same data with the two routines, instead of acquiring again. Though this would be the systematic way to proceed, the data seem stable enough across the two acquisitions (at least visually) that the refined procedure may be just a waste of time. Especially because it's not simple to achieve with a runcard, and currently non-trivial even with a script -> but it's an idea to improve scripts capabilities!

Edoardo-Pedicillo, 2024-10-28:
> However, to complete the empirical proof (i.e. attribute the responsibility for this specific case), we should actually fit the same data with the two routines, instead of acquiring again.

I tried with the data in http://login.qrccluster.com:9000/qYLzOD1ET9yCXMAr0ZsNmA==/#T2 
This is the fit results 
[image]

and this the plot of the differenze between the T2 and Ramsey fit

[image]

So as @alecandido pointed out we have the sinusoidal contamination.

> However, in some cases even if you correct the qubit frequency T2 can exhibit a non monotonic behavior due to some spurious qubit-qubit coupling.

Or it could depends on the type of noise 

[image]

> At this point we can try to pre-process the data before doing the fit in order to choose whether we should perform an exponential fit or an exponential fit + sinusoidal term. On the top my head I'm [...]

I am not sure this could work in case the Ramsey oscillations have a period  comparable with or higher than T2

Edoardo-Pedicillo, 2024-10-28:
And in the following plot I am removing 0.5 from the two fits and evaluating the ratio, this should give the sinusoidal fitted by the ramsey, that seems to be just noise.

[image]

alecandido, 2024-10-28:
Thanks @Edoardo-Pedicillo, that's very interesting.

For the time being (based on this data), I agree that preprocessing mostly won't make a huge difference, since it's not much the fit initialization the problem, but rather the landscape around the minimum.

So, yes, we may want to have a criterion to fit the pure exponential, or fall back on the sine addition. But based on these data, I'm not even sure it has always to be a sine, because spurious terms may come from sources other than a residual detuning.
In any case, there's quite a chance the best criterion would be some measurement of "failure" of the exponential fit itself.

alecandido, 2024-10-28:
> And in the following plot I am removing 0.5 from the two fits and evaluating the ratio, this should give the sinusoidal fitted by the ramsey, that seems to be just noise.

It is not exactly the sine, but (assuming 0.5 is the value of the fitted constant), it will be exactly

$$
1 + \frac{\sin(\dots)}{\exp(\dots)}
$$

and indeed, you see that the right tail is enhanced by the exponential getting smaller (you're dividing the sum of exp and sine, the Ramsey fit, by just the exp, the T2).

Edoardo-Pedicillo, 2024-10-29:
> (you're dividing the sum of exp and sine, the Ramsey fit, by just the exp, the T2 - assuming the exp is the same).

The Ramsey fit is the product of the two.

To sum up, a potential solution can be outlined in the following workflow:
- Evaluate the T2 with Ramsey routine
- If the T2 error is too big execute the T2 experiment

This process can be easily implemented in a script. Additionally, another interesting approach to explore was suggested by @andrea-pasquale (I plan to open an issue regarding this).

> At this point we can try to pre-process the data before doing the fit in order to choose whether we should perform an exponential fit or an exponential fit + sinusoidal term. On the top my head I'm [...]

Theoretically, it should work since the FFT of the Ramsey fit function has a peak in correspondence of the oscillation frequency, but we need more evidences about the robustness before deploying.

closed 2024-10-29 by Edoardo-Pedicillo

alecandido, 2024-10-29:
> The Ramsey fit is the product of the two.

Correct, I was wrong about that. Indeed, the enhancement in the oscillations is most likely due to the difference in the fitted decay rates (in https://github.com/qiboteam/qibocal/issues/1023#issuecomment-2437245308 a 10% difference was reported).

Moreover, I'd advise the opposite: evaluate T2 with the T2 experiment, and resort to the Ramsey fit only if failing (since you may even get a slightly wrong central value, and the error is not guaranteed to work as a reliable marker). But I will comment about this in the new issue.

Edoardo-Pedicillo, 2024-10-30:
> Moreover, I'd advise the opposite: evaluate T2 with the T2 experiment, and resort to the Ramsey fit only if failing (since you may even get a slightly wrong central value, and the error is not guaranteed to work as a reliable marker).

This approach is not robust enough as already pointed out here 

> In general you can always say something like "I will run with detuning and afterwards I will run just the T2 experiment to measure T2". However, in some cases even if you correct the qubit frequency T2 can exhibit a non monotonic behavior due to some spurious qubit-qubit coupling.

So it is more likely that you have to execute two fits.

For what we have seen, it is better to do as I said in the previous message. Of course this is a suggestion, we are enough flexible and everyone can do whatever he wants :)

alecandido, 2024-10-30:
> This approach is not robust enough as already pointed out here

I honestly do not understand the argument...

Here you showed that, for this specific case, if you just started from the pure exponential fit you would have stopped there, because it just worked.
Then, I agree this may not always be the case, but I'm not seeing where the approach is failing... 

> So it is more likely that you have to execute two fits.

well, not in this case, for example...

Moreover, the proposed alternative always requires two fits (the FFT peak identification, and the exponential w/ or w/o sine).

> For what we have seen, it is better to do as I said in the previous message. Of course this is a suggestion, we are enough flexible and everyone can do whatever he wants :)

Well, it's not much about proving who is right, or imposing everyone a single strategy without any flexibility.
But it's convenient to dig deeper and compare the alternatives, truly estimating the pros and cons and getting as much as possible to a measurement of the "robustness" of the two, since this is what is making us closer to automation.
For specific cases, you may need a different strategy. But it'd be much better to identify an overall more robust one, or instead to discriminate (even partially) when it seems more convenient to apply one or the other.