qiboteam/qibocal #1037: Add experiment for qubit frequency vs flux amplitude

andrea-pasquale, 2024-11-05:
In the past we had several discussion about the possibility of sweeping offset or flux amplitude in the [flux dependence ](https://qibo.science/qibocal/stable/protocols/flux/single.html) experiment. I believe that the current solution (sweeping offset) is correct given that from the protocol we aim at extracting the sweetspot of the qubit which will be set as the offset of the device.

However, especially when dealing with two qubit gates, it is interesting to derive also how the frequency of the qubit changes when applying a flux pulse at fixed amplitude for a given duration. Although we could infer this from knowing how the qubit frequency changes when we modify the offset, there is no clear relation between the voltage generated by the offset and how it is modified when we apply a flux pulse. The relationship is expected to be linear $V = \Phi + C A$ and in many cases $C=1$, but it is not known before running the experiment.

I propose to add an experiment that specifically tackles the qubit frequency vs flux amplitude dependency. To avoid overlapping with the previous experiment, the new experiment will be more suited for short flux pulses. A prototype for this experiment is available [here](https://github.com/qiboteam/qibocal/blob/61513b32eae1edcdfd1ad746ecb70f3e95cb4a97/src/qibocal/protocols/characterization/two_qubit_interaction/cryoscope_amplitude.py), which consists of applying a flux pulse for qubit in a superposition state and then we perform GST to measure the expecation values of X and Y to compute the detuning on the qubit frequency.

As long as we tackle small detunings a simple parabola fit could do the job. We can also store the parameters of the fit in the new `Qubit` object available in #1021 

```python
class Qubit(Model):
    """Representation of Qubit parameters"""

    ...
    flux_amplitude_coeff: Optional[list] = None

    def detuning(self, amplitude: float) -> float:
        return self.flux_amplitude_coeff[0]*amplitude**2 ...
```
We could even complicate the function by also considering the flux offset.

```py
def expected_frequency(self, offset: float, amplitude: float):
    return frequency(offset, ...) + self.detuning(amplitude)
```

The two previous examples are just a possible way of using the newly found parameters to predict the qubit frequency.

Thoughts @Edoardo-Pedicillo @hay-k?

alecandido, 2024-11-09:
Why not performing the exact same qubit flux dependency experiment, fitting the same transmon shape, but instead sweeping the flux amplitude about the offset?

If you expect it to be just a constant rescaling, you should be able to just fit it in the exact same way, without any GST or anything different. Am I missing something?

andrea-pasquale, 2024-11-10:
> Why not performing the exact same qubit flux dependency experiment, fitting the same transmon shape, but instead sweeping the flux amplitude about the offset?

In theory we could, and it was already done in #691, where we were sending a long flux pulse and we were perfoming the measurement withing the flux pulse. However, in common applications, such as two-qubit gates calibration experiments, this will not be the expected scenario. We usually send a short flux pulse to bias the qubit/coupler away from its sweetspot and we then perform other operations. Since the pulse is short we cannot perform the readout within the duration of the flux pulse.
For this reason the preferred approach to measure the detuning caused by short flux pulses is to perform GST.

Measuring with GST should also allows us to have a better understanding of what is happening with virtual phases.

alecandido, 2024-11-11:
Ok, I see your point.

However, I expect that the measurement of the $C$ value above can differ from 1 because of physical effects due to the electronics supposed to do the pulse addition on top of the offset. Which is non-ideal, and can thus generate a non-trivial effect, which may even depend on the offset itself (i.e. $C \equiv C(\Phi)$, but we'd measure it at the offset used, so we're just interested in a single value).
In this picture, there is no relation between the $C$ value and the duration of the pulse, since you're just assuming the addition to be non-ideal by introducing a further factor.

Of course, this factor may even depend on the pulse shape, i.e. the duration. But at that point, I see no reason it should not depend even on the pulse amplitude itself, making it $C \equiv C(\Phi, A, \Delta t)$, in which case there would be no linearity on $A$. You can still assume deviations from linearity to be small. But if they are relevant, they can not be considered small.
In this picture, in general you're not calibrating a coefficient, but an entire function:

$$
V = \Phi + f(\Phi, \mathcal{P})
$$

where $\mathcal{P}$ would be the pulse (the whole shape).

The point I'm getting to is that, if short flux pulses have to be considered different from long ones, doesn't that calibration overlap with the more generic pre-distortions?

If there is a *standard approach* there will also be a motivation - which I'm failing to so, because my lack of expertise. So, if you can point out references, I can have a look by myself (but if you have the full explanation as well, that would also be appreciated, of course).

closed 2025-03-13 by andrea-pasquale