qiboteam/qibocal #1054: Enhancements to DRAG protocol

andrea-pasquale, 2024-12-06:
Currently our DRAG calibration is checking for phase errors using a sequence with $X_\pi - X_{-\pi}$.
Similar to the flipping protocol it is possible to apply a sequence which is $(X_\pi - X_{-\pi})^N$ which should generate more clearly oscillations. Both [qua-libs](https://github.com/qua-platform/qua-libs/tree/a1225b95da7a8f760d1abd55b1e76ea65ba07e03/Quantum-Control-Applications/Superconducting/Single-Flux-Tunable-Transmon/Use%20Case%202%20-%20DRAG%20coefficient%20calibration) and qiskit-[experiments](https://qiskit-community.github.io/qiskit-experiments/stubs/qiskit_experiments.library.characterization.RoughDrag.html) allow to do this.

Moreover, if we want something which doesn't involve oscillation there is another way of selecting the value of $\beta$ which involves testing just two sequences which should highlight phase errors. There is an example of this procedure in [qua-libs](https://github.com/qua-platform/qua-libs/tree/a1225b95da7a8f760d1abd55b1e76ea65ba07e03/Quantum-Control-Applications/Superconducting/Single-Flux-Tunable-Transmon/Use%20Case%202%20-%20DRAG%20coefficient%20calibration) again. This second fitting procedure was first tested at Yale, see for example https://rsl.yale.edu/sites/default/files/2024-08/2013-RSL-Thesis-Matthew-Reed.pdf.

Finally, we have already mentioned a few times that after finding the correct value for $\beta$ in order to address both leakage and phase errors it is possible to use a waveform where we introduce a detuning

$$
s(t) \rightarrow e^{i \delta \omega t} s(t)
$$
which is explained [here](https://arxiv.org/abs/1904.06560) and also [here](https://arxiv.org/pdf/1509.05470).
In this case we should add this waveform to qibolab.

alecandido, 2024-12-06:
> In this case we should add this waveform to qibolab.

Not sure what you mean exactly, but we can definitely modify waveforms, if needed.

I would say that in this case, adding a detuning just means changing the modulation frequency, so it has no effect on the envelope (you can certainly pre-modulate the envelope with a small detuning, but it seems a bit impractical...).

andrea-pasquale, 2024-12-06:
Perhaps it is better to discuss offline, according to the references which I've provided above I think that they are using a different waveform since they say "By introducing a frequency detuning parameter $\delta f$ to the waveform". Moreover, according to the data provided by them the detuning could be up to a few tens of MHz which it will probably not work if we use it as demodulation frequency.