qiboteam/qibocal #1067: Qubit flux - sweetspot in the results file is not within the range of the bias width of the measurement.

Luca-Ben-Herrmann, 2025-01-22:
When running a qubit flux measurement with specified bias width the sweetspot that in the results folder is not the one that corresponds to the fit in the specified bias region. I am assuming the reason for this is that the sweetspot that is updated in the results file is not constrained to the width of the sweep, but rather it extrapolates to the entire bias range of the device and takes the sweetspot that is closer to 0. @andrea-pasquale, @Edoardo-Pedicillo, @hay-k What do you think? could we fix for example by making sure that the sweetspot that is updated in the results file must be in the range of the bias width of the measurement?

See the following report (specifically look at the qubit flux measurement of qubit 3 )

andrea-pasquale, 2025-01-22:
Thanks @Luca-Ben-Herrmann for opening the issue. I cannot see the report. Perhaps it was not uploaded correctly to GitHub.

Luca-Ben-Herrmann, 2025-01-22:
I edited the message, and it includes the link to the report now @andrea-pasquale.

Edoardo-Pedicillo, 2025-01-22:
> I am assuming the reason for this is that the sweetspot that is updated in the results file is not constrained to the width of the sweep, but rather it extrapolates to the entire bias range of the device and takes the sweetspot that is closer to 0.

Thanks @Luca-Ben-Herrmann for the issue. This is exactly what it is happening and is expected. https://github.com/qiboteam/qibocal/blob/293e2a9105568026bb2ee2a94062b381aa8c34bc/src/qibocal/protocols/flux_dependence/qubit_flux_dependence.py#L221-L223 We could return the sweetspot closer to the center of the window that we are sweeping, or for simpicity the first positive value of the sweetspot.

andrea-pasquale, 2025-01-24:
Fixed in #1068

closed 2025-01-24 by andrea-pasquale