qiboteam/qibocal #1075: Circuit based protocols use default native gate set instead of platform native gates

jykhoo1987, 2025-02-05:
Circuit based protocols (e.g. 1 and 2 qubit rb) uses the default native gate set when calling the transpiler:

cls.CZ | cls.GPI2 | cls.I | cls.Z | cls.RZ | cls.M

When the platform has a different native gate set, these protocols cannot execute.

These protocols should allow for native gates to be declared in the qibocal runcard as a protocol parameter (such as how it is done in chevron), or use the native gates declared in the qibolab runcard instead of the above native gate set.

alecandido, 2025-02-05:
Just to clarify the situation about Qibolab 0.1/0.2: the default hardware transpiler is making use of the platform natives since Qibo 0.2.4
https://github.com/qiboteam/qibo/blob/bbf0ccf366106a982d68f8e8881010c4c9272707/src/qibo/backends/__init__.py#L139

If used, this should solve this issue on its own.

In principle, the mechanism has been conceived to be compatible with all Qibolab releases. However, the 0.2 support has never been merged
https://github.com/qiboteam/qibolab/pull/1083

Whenever needed, we (i.e. me and @stavros11) will take over that PR and complete the support. It will be a reasonable task taking a reasonable amount of time :)

Edoardo-Pedicillo, 2025-02-05:
At the time being, Qibocal is not using the default transpiler, because we wanted to be detached from the latest transpiler refactoring. So the issue is internal to the Qibocal library.
https://github.com/qiboteam/qibocal/blob/bc7a7badcfa4a75f2c20d179c509ab2d823ce13b/src/qibocal/auto/transpile.py#L113 

In the future we can consider the idea of using the default hardware transpiler, but at the moment I will open a PR with a patch for this problem.

closed 2025-02-14 by Edoardo-Pedicillo