qiboteam/qibocal #1088: Error with `chevron` acquisition

ElStabilini, 2025-02-15:
During the past week, I have faced the following problem multiple times while trying to run the `chevron` protocol.

The runcard I tried to run is the following:
```
platform: qw11q

targets: [ [B2, B4] ]

actions:

  - id: chevron CZ

    operation: chevron
    parameters:
      amplitude_min: -0.49
      amplitude_max: -0.43
      amplitude_step: 0.002
      duration_max: 120
      duration_min: 0
      duration_step: 1
      native: CZ
      nshots: 256
      relaxation_time: 100000

```

I was working working on commit 0f0e582c of `qibolab`, branch `test_cryoscope_postprocessing` (which points to 0.2) of `qibocal`, and a `qibolab_platform_qrc` branch with fir filters obtained with cryoscope and pointing to `b_line_rererererecal-0.2`.

The protocol fails during acquisition with the output in the attached file.

[slurm-734018.txt](https://github.com/user-attachments/files/18809624/slurm-734018.txt)

As @alecandido suggested I report here also the following objects (printed also in the slurm output`)

- `ro_low.id`:  `139818938987152`
- `results`: `{139818938987152: None, 139818938987472: None}`

Maybe it is also useful to say that the acquisition works correctly at least in the amplitude interval [0.5, -0,2) which I completely scanned by performing multiple chevron protocols on sub-intervals.
 
My get-around for the problem was to move to the most recent commit of `qibolab`.

ElStabilini, 2025-02-15:
For completeness, I add also the output of `pip freeze`. 
I realized that the first time I saw the error I do not remember whether I ran the `poetry install` command for `qibocal` or `qibolab` first.

[env.txt](https://github.com/user-attachments/files/18809989/env.txt)
[env1.txt](https://github.com/user-attachments/files/18809988/env1.txt)

In either case, I was able to reproduce the same error with both the resulting environments

Edoardo-Pedicillo, 2025-02-17:
Thanks @ElStabilini for the reporting. The issue seems not related with `Qibocal`. I have tried with both https://github.com/qiboteam/qibolab/commit/0f0e582c and https://github.com/qiboteam/qibolab/commit/74d98c61c1328647359229951a13e839a0143142 (the actual `main`) and I still see the issue. After some investigations, it seems the issue is coming from https://github.com/qiboteam/qibolab/commit/0bb74e014c46b743a035303a7a7163b158737e8a. @stavros11 Could you have a look? At the moment `v0.2.2` it's working.

alecandido, 2025-02-17:
It seems clear to me that the problem is not related to Qibocal, since `platform.execute()` should never return `None` values. I'm transferring this to Qibolab