qiboteam/qibocal #1091: Issues installing `qibocal`

renatomello, 2025-02-17:
1. Running `pip install qibocal` install version `0.0.1`
2. The above command also tries to install `pandas 1.53` from source, which slow downs the installation by a big margin
3. `qibocal`and `qibolab` are not working together when both are installed from their latest versions
4. Installing both from `main` branch also does not work
5. `qibocal` currently doesn't support `python 3.12`
6. to get versions of `qibocal` and `qibolab` that talk to each other, one needs to downgrade to `python 3.11`

renatomello, 2025-02-17:
@scarrazza @alecandido

alecandido, 2025-02-17:
> 1. Running `pip install qibocal` install version `0.0.1`

I did not manage to reproduce this. To me, it is correctly installing v0.1.2, as it results from PyPI
https://pypi.org/project/qibocal/

> 2. The above command also tries to install `pandas 1.53` from source, which slow downs the installation by a big margin

I acknowledge it's annoying. But it should not happen with newer versions.

> 3. `qibocal`and `qibolab` are not working together when both are installed from their latest versions
> 4. Installing both from `main` branch also does not work

Unfortunately, this is because there is an ongoing transition, which should be mainly resolved in these days.
We have two active lines of development of Qibolab (0.1 maintenance, and 0.2 development). This is typical for large projects (such as Python itself, cf. https://devguide.python.org/versions/). We do not want to waste resources in maintaining both. But it has been quite a huge transition, and it took time for Qibocal to catch up. Now it is largely done, and #990 will be merged very soon.

In any case, the result is that any Qibolab 0.2 version (including `main`) should be matched with Qibocal #990, while if you want to use Qibocal releases or its present `main` (soon to become the `0.1` branch, or `qibolab0.1`), you can, just installing a Qibolab 0.1 release (or Qibolab's `0.1` branch itself).
The version specifier of Qibocal should do exactly that
https://github.com/qiboteam/qibocal/blob/7a25e48b442ecddfece20567b32b80fd5f584bf8/pyproject.toml#L20

alecandido, 2025-02-17:
Sorry, I left two points behind.

> 5. `qibocal` currently doesn't support `python 3.12`
> 6. to get versions of `qibocal` and `qibolab` that talk to each other, one needs to downgrade to `python 3.11`

Unfortunately, this is the result of an optional dependency of Qibolab, namely the QM package itself (which is currently the main driver used).

We are not writing code which is specific to any Python version, and we would be broadly compatible with all Python versions (we actually target only active ones, and we use features introduced in the oldest maintained version or before, not newer ones - currently we stick to py3.9 for this reason).
However, we would have no forward bound, for as long as Python does not break the compatibility on its side (which mostly does not happen - though some deprecation and removal have been enacted, but we were not using any of the removed modules).

Thus, forward bounds are fully a consequence of our dependencies.

Only recently, QM released an official py3.12 compatible version. And Qibolab propagated the support. Qibocal `main` did not, but #990 already supports py3.12.
https://github.com/qiboteam/qibocal/blob/6d7f8a6cc94f6db67e4436874a95d14a8fa17cc9/pyproject.toml#L19

So, this problem is already solved, and it will be soon solved even in `main` and a release.

Sorry for the temporary troubles.

renatomello, 2025-02-17:
Thanks for the answers. I finally managed to install `qibolab` from `main` and `qibocal` from the `0.2` branch. However, the dependencies `qcodes`, `qm-qua`, and `qm-octave` needed to be installed to make it work even though they are set as optional in the `pyproject.toml`

alecandido, 2025-02-17:
> However, the dependencies `qcodes`, `qm-qua`, and `qm-octave` needed to be installed to make it work even though they are set as optional in the `pyproject.toml`

Indeed. This is quite a pain point, for which I never found a pleasant solution...
```py
pip install qibocal
```
will also install `qibolab` (even the proper version, we usually make sure the dependency to be correctly synced), but not its optional dependency.
Then, if you wish to run on hardware, you need to run, e.g.
```py
pip install qibolab[qm,los]==<version-installed-by-qibocal>
```
to get the other dependencies you're missing (without altering the version resolved - which I hope it won't be much needed any longer, since we aim for broad compatibility).

However, propagating all optional dependencies to Qibocal seemed exceedingly redundant.

I'm considering to distribute some empty packages `qibolab-qm`, `qibolab-qblox`, ... to alleviate the problem. In that case, you should only install
```py
pip install qibocal,qibolab-qm,qibolab-los
```
or something like that. The `qibolab` dependency itself will be provided as a `qibocal` dependency, while the further ones will only take care of the extra dependencies (and no code on their own). But there would be still a problem syncing them with the various versions (if needed).

Possibly, the best solution will be to stabilize even the drivers' interface, and distribute the drivers as independent packages, and not as part of the main `qibolab` package (which will just define the interface).

For the time being, I considered all of this excessive overhead, and postponed the issue of the optional dependencies. But I acknowledge that the environment setup is less than optimal, and I keep thinking about alternatives (I'm also very open to proposals).

Edoardo-Pedicillo, 2025-02-18:
> 1. Running `pip install qibocal` install version `0.0.1`

I managed to reproduce the issue only with python 3.12 . Indeed, for 0.0.1 we have https://github.com/qiboteam/qibocal/blob/581f1de22e7a7503b41413a3f1b5d30112663e65/pyproject.toml#L18

In all the other versions, the constraints for python are the usual ">=3.9,<3.12".

alecandido, 2025-02-18:
You're right, I tried with py3.11. That's definitely the explanation.

Then, it's still part of what has been described in https://github.com/qiboteam/qibocal/issues/1091#issuecomment-2663671652, and stemming from Qibolab 0.1 still not supporting py3.12.
We could even update all the 0.1 bundle, since the `qm-qua` dependency in there is already v1.2.1, and so supporting py3.12. But given that 0.2 is finally rolling out, I'm not sure how much is relevant.

In any case, thanks @Edoardo-Pedicillo!

closed 2025-05-05 by Edoardo-Pedicillo