qiboteam/qibocal #1096: Check amplitude less than 1 in _fit

Edoardo-Pedicillo, 2025-02-20:
As the title says.

alecandido, 2025-02-20:
Would you elaborate on the motivation or the specific routines which you have in mind?

Pulses' amplitudes are always enforced to be less than 1 in Qibolab (thus in `_acquire`), and I guess most fits are looking for values within the scanned range. But I may be wrong.

Edoardo-Pedicillo, 2025-02-20:
> I guess most fits are looking for values within the scanned range. But I may be wrong.

No, the fit has some bounds on the normalized paramters, like in rabi https://github.com/qiboteam/qibocal/blob/5520bba557c5a116e4eb0bc2186a2f52e45f2755/src/qibocal/protocols/rabi/utils.py#L348-L373 
because in 0.1 it was not necessary.

RoyStegeman, 2026-10-31:
Probably better to check that amplitude values are less than the largest amplitude included in the sweep. I.e. check that the fitted "best amplitude" is not extrapolated. Then don't update the parameter if this condition is not met, and potentially log a warning explaining the situation and a possible solution (increase pulse duration or reduce attenuation).

alecandido, 2026-11-02:
I'm not even sure if we want to be strictly within the range, or impose the condition with a certain tolerance (e.g. 20%, or anything like that).

If it is far enough, we are just guessing in the wild, at some point it is hard to resolve the correct frequency of the oscillation. Technically, if the measurement is super-accurate, we could a fit from whatever arbitrary small neighborhood, just because a sine is analytic. But that's just abstract, and convergence in random local minima may screw any "distance" assessment based on the error reported by the fit.
Given these considerations, assuming that we are not extrapolating is just safe, and sane. Or, at least, not "far" extrapolating. Which can be tuned, with a further hyperpar.