qiboteam/qibocal #1146: Qubit Flux fitting fails often

Luca-Ben-Herrmann, 2025-05-15:
The fit for the qubit flux measurement typically fails. For example, see the following report: http://login.qrccluster.com:9000/-M3x03X8TPS9sIBN9hwUyw==/ (specifically for qubits 1 and 3 in this report I would expect the fit to work since the contrast is sufficient).

Additionally (lower priority) depending on the readout resonator frequency chosen, we sometimes have better contrast in the "phase" than in the "signal". Perhaps it would be possible to have as a parameter of the qubit flux routine, whether we want to fit the signal or phase. Any opinions/suggestions are welcome!

andrea-pasquale, 2025-05-15:
Thanks for opening the issue @Luca-Ben-Herrmann.
Probably the failure of the fit is related to this:
https://github.com/qiboteam/qibocal/blob/b7455657dad44d8b9b9deb76823112ed6d819b0d/src/qibocal/protocols/flux_dependence/qubit_flux_dependence.py#L204-L209
Before performing the fit we select only the relevant points by applying a mask through the function `extract_feature`. This function will apply a mask that select only the points corresponding to a larger signal value when passing `max` and the points with lower signal value when passing `min`. Since this is a qubit spectroscopy and this chip has a `2D` resonator the fit is looking for some peaks in the spectrum, which is not the case for qubits 1 and 3. This is probably due to the fact that the readout frequency is not exactly at the dressed resonator frequency (which could be the case if before you ran some readout optimization protocol). 
Here is the fit when the appropriate mask is used: http://login.qrccluster.com:9000/gDLN80PlQdOu6bIjtdo6IQ==
To avoid such errors we have a few solutions:
1- Let the fit figure out whether to select the points corresponding to peaks or dips
2 - Make sure that when we do experiments not involving probabilities we measure at exactly the dressed resonator frequency
3- Fit using also the information in the phase
In https://github.com/qiboteam/qibocal/pull/1071 I tried something along the direction of 2, basically for signal experiments we made sure to readout using the dressed resonator frequency.
https://github.com/qiboteam/qibocal/blob/0d9eb042cf4f539424015fa94ce4730a256fd480/src/qibocal/protocols/flux_dependence/qubit_flux_dependence.py#L158-L161
This parameter (the dressed resonator frequency) can be set manually in the `calibration.json` by updating the platform after running either the `punchout` or the `dispersive_shift` experiments.

alecandido, 2025-05-16:
@andrea-pasquale's answer is complete enough, and raising the correct points to be discussed, at least in the short term.

Just to mention, since relevant to the topic discussed, I opened some time ago an issue related to a possible reassessment of the `extract_feature()` strategy, https://github.com/qiboteam/qibocal/issues/956
It's not something you should consider for a quick solution, but it may be useful if at some point you could review even that and push the discussion (that could end up in the proposal dismissal, but at least the motivation will be documented somewhere)

Luca-Ben-Herrmann, 2025-05-20:
@andrea-pasquale thanks a lot for the elaborate explanation!

I tried option 2 (using the dressed frequency in the calibration.json) and it works very well!

Regarding the other two options, I think that they would definitely be useful (also for other measurements), but for now since option 2 is working and is the easiest solution I think I can close this issue.

closed 2025-05-20 by Luca-Ben-Herrmann