qiboteam/qibocal #1148: Allxy drag pulse tuning routine missing on 0.2

Luca-Ben-Herrmann, 2025-05-15:
The allxy drag pulse tuning routine is missing on 0.2 (it exists on 0.1). The routine would be useful to further optimize the drag pulse, but since we have other routine to do that it is not high priority at the moment (at least for me).

andrea-pasquale, 2025-05-15:
The reason why we decided to sto supporting the drag pulse tuning routine is that it is merely a repetition of the allxy experiments with different values of the drag parameter, which as you said can be extracted using other protocols.
If the same plot is necessary a possible solution is to run singularly experiments for different values of the drag parameter and then combine them using `qq compare`.

alecandido, 2025-05-20:
Agreed that software loops should be disfavored wrt sweeps.

At a practical level, there would be no way to sweep envelopes-specific parameters with Qibolab right now. A limitation we can (and will) address, but it will also require some time to deliver.

All in all, I also see the point in not proliferating the routines, and have a better composition. But that will call a trade-off about not proliferating routines' parameters, and overcomplicating the various steps (acquisition, fit, plot).
When it comes to operations you can not do in real-time, it is worth to avoid the overhead of a further routine, and just wrap the execution of multiple ones in a single script. However, if you need real-time operation, that it is better to modularize the routine components (e.g. the sequence definition as an individual function), and reuse in multiple routines (as already being done for the various Rabi and similar).

> That being said, it is not top priority for now since we also have other routines to tune the drag pulse.

For the time being, if that's confirmed, I would put this issue behind the proper support in Qibolab for the required sweeps. About which, I'm just going to write a dedicated issue.