qiboteam/qibocal #1149: Cryoscope routine: number of maximum FIR taps smaller than what QM provides

Luca-Ben-Herrmann, 2025-05-15:
With the current version of the cryoscope routine the maximum number of FIR taps that can be used is 30. According to the QM (https://docs.quantum-machines.co/1.2.3/docs/Guides/output_filter/?h=filters#using-the-filters) it should be possible to use 48 taps for the FIR filter.

andrea-pasquale, 2025-05-15:
Thanks for opening the issue.
At least from the qibocal side I believe that we didn't put anywhere a limit on the number of FIR taps, perhaps it could be related to the specific QM device that you are using. Do you have an example where you tried to use more than 30 FIR and the code failed?

Luca-Ben-Herrmann, 2025-05-16:
When @Edoardo-Pedicillo and I were working on it we observed the following: 
1. Whenever we use more than 38 taps we get an error saying that the maximum number of taps is 38.
2. When using 31 to 38 taps we got an error saying that the maximum number of taps is 30.
3. When using 30 or less FIR taps it works. 

We don’t have the logs saved but we can try to reproduce the errors when the fridge is cold again. 

I agree that it is most likely an error related to the QM. Perhaps it will be resolved with the new QOP update. I will keep you updated as soon as we test it again.