qiboteam/qibocal #1161: RB report writing fail

sorewachigauyo, 2025-05-23:
When trying out the standard RB on Qibocal 0.2.1, I get the following error

```python
Traceback (most recent call last):
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/bin/qq", line 8, in <module>
    sys.exit(command())
             ^^^^^^^^^
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/click/core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/qibocal/cli/_base.py", line 77, in run
    protocols_execution(runcard, folder, force, update)
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/qibocal/cli/run.py", line 61, in protocols_execution
    report(path, history)
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/site-packages/qibocal/cli/report.py", line 96, in report
    (path / "index.html").write_text(html)
  File "/mnt/home/nqch_paul/.conda/envs/sinq-20-dev/lib/python3.11/pathlib.py", line 1079, in write_text
    return f.write(data)
           ^^^^^^^^^^^^^
UnicodeEncodeError: 'ascii' codec can't encode character '\xb1' in position 11166: ordinal not in range(128)
```

Test runcard
```yml
platform: sinq-20
targets: [19]

actions:
- id: rb
  operation: standard_rb
  parameters:
    depths:
      start: 1
      stop: 10
      step: 1
    niter: 1
    nshots: 200
```

[rb.tar.gz](https://github.com/user-attachments/files/20407381/rb.tar.gz)

alecandido, 2025-05-23:
Not sure why `ascii` is the defaulting encoding instead of `utf-8` on your platform, but that's clearly the problem.

```py
[ins] In [1]: "\xb1"
Out[1]: '±'
```

That is for sure in some tables...

[`Path.write_text()`](https://docs.python.org/3/library/pathlib.html#pathlib.Path.write_text) used here refers to [`open()`](https://docs.python.org/3/library/functions.html#open) built-in function.

> *encoding* is the name of the encoding used to decode or encode the file. This should only be used in text mode. The default encoding is platform dependent (whatever [locale.getencoding()](https://docs.python.org/3/library/locale.html#locale.getencoding) returns), but any [text encoding](https://docs.python.org/3/glossary.html#term-text-encoding) supported by Python can be used. See the [codecs](https://docs.python.org/3/library/codecs.html#module-codecs) module for the list of supported encodings.

This could be solved by always specifying the encoding while writing, fixing it to `utf-8`. However, you could also solve on your own just changing the locale.

alecandido, 2025-05-23:
According to [Python UTF-8 Mode](https://docs.python.org/3/library/os.html#utf8-mode) you could also just set `PYTHONIOENCODING` if you do not want to override your system encoding

sorewachigauyo, 2025-05-24:
Yeah, works fine outside of Qibocal

```python
>>> import pathlib
>>> pathlib.Path("ciao").write_text("\xb1")
1
>>> pathlib.Path("ciao").read_text()
'±'
```

closed 2025-05-28 by andrea-pasquale