qiboteam/qibocal #1171: Generalize qq upload

scarrazza, 2025-06-04:
We should add support to custom host configuration so other labs can adjust their environment and use `qq upload` e.g. by exposing environment variables such as `UPLOAD_HOST`, `TARGET_DIR`, `ROOT_URL` as defined [here](https://github.com/qiboteam/qibocal/blob/main/src/qibocal/cli/upload.py#L17).

andrea-pasquale, 2025-06-04:
That is reasonable. Would it be enough for now to expose them as CLI arguments for now?

alecandido, 2025-06-04:
Maybe we should even support a configuration file, such that we could wipe the defaults out of Qibocal, and just dump them on a file somewhere.

Since conventions may vary on Linux vs Windows, you may even extract defaults from third-party packages, e.g. [`platformdirs`](https://github.com/tox-dev/platformdirs) (a further dependency, but only for the CLI).