qiboteam/qibocal #1174: Rabi signal experiments use readout frequency from `calibration.json` and not `parameters.json`

sorewachigauyo, 2025-06-05:
In [`rabi_length_signal`](https://github.com/qiboteam/qibocal/blob/main/src/qibocal/protocols/rabi/length_signal.py#L107-L110), the probe channel frequency is set from the calibration. Normally, this would not be a problem as the channel config should be the same as the calibration from the resonator spectroscopies, but I was unable to run them due to driver issues.

Is this behavior expected?

andrea-pasquale, 2025-06-05:
Yes, this is expected. We decided to go for this approach since if the readout frequency is optimized to get better fidelities its value might not be the value of the dressed resonator frequency which is the optimal value for getting something meaningful during "signal" experiments. 

Probably we could document this better.

RoyStegeman, 2026-11-02:
The question was answered. There is still the relevant observation that "Probably we could document this better.", but as this holds for most of the documentation I don't think we need an issue issue to track this specific scenario

closed 2026-11-02 by RoyStegeman