qiboteam/qibocal #1206: Compute QND using the implementation in `readout_characterization`

ElStabilini, 2025-07-21:
Since the QND calculation is already available in the `readout_characterization` protocol, it could be useful to use this function in the `resonator_optimization` protocol instead of directly computing QND.

Thank you @andrea-pasquale for the suggestion.

ElStabilini, 2025-07-21:
Related to the QND calculation in the `resonator_optimization` protocol, it may also be interesting to investigate the reason for the observed pattern of both QND and QND-pi as a function of the amplitude and frequency of the readout resonator.

What we saw from report is similar to the following image

[image]
[report](http://login.qrccluster.com:9000/brbo8k-_RUmeDOraTnI8ZQ==/)

while we expected a plot  similar to the fidelity one.

RoyStegeman, 2026-11-02:
closed by https://github.com/qiboteam/qibocal/pull/1211

closed 2026-11-02 by RoyStegeman